wireviz/WireViz #12: Multiple colors per wire

viperdream26, 2020-06-24:
Is it possible to have multiple colors per wire? If not possible yet, it would be a huge improvement. Thanks for all the hard work!

aakatz3, 2020-06-24:
Out of curiosity, would this be something like striped/banded wires, like in [CAT-5](http://wiringwizard.com/primer/cables/cat5/rj45.gif), or the [25 color code](https://en.wikipedia.org/wiki/25-pair_color_code#25-pair_telco_cable_pinout)?

Or, do you mean something like a red wire becomes a white wire as it's daisy-chained along?

viperdream26, 2020-06-24:
@aakatz3 I meant striped/banded wires, which are incredibly common.

You bring up another interesting point there, but I don't know how common that is?

aakatz3, 2020-06-24:
Striped/banded should theoretically be easy, I do need to look into the code a bit to see if a pattern is possible, and then maybe you can specify a primary/secondary.

While it shouldn't be common, I've seen it in a car harness once or twice, and I can imagine a formula SAE team may want that feature, but it's probably bad practice.

chrisalbertson, 2020-06-24:
I certainly make cables using striped wire.

I try really hard to never do the second thing and change wire color but
just recently I had to.  I have motors that have wires with red, black,
white, yellow, and these needed to be extended with a cable that had red,
black, white, BLUE.    I hate splicing cables where the colors don't match
but some times there is no other option.  It would be nice to be able to
document this.

More:

   - It would be nice if there were a way to capture the length of a cable.

   - Also some connector shells allow mixed genders where each pin can be M
   or F.
   - Then there are the "genderless" type that mate with themselves

holgua, 2020-07-19:
> @aakatz3 I meant striped/banded wires, which are incredibly common.

I would love to have that feature as well. In vintage automotive wiring it is common to use striped wires (green-white, white-green, ...). Without it, it's impossible to tell circuits apart.

e.g. temperature gauge hooked up via red-white wire here https://www.mre-books.com/electric/images/wiring1.jpg

formatc1702, 2020-07-19:
#96 has been merged into `dev` and will be part of v0.2

PR #96 (2020-07-17): Implement Feature/multicolor wires on refactored code
This is a rewrite/merge for #17, after a development reset. I decided to do this in a new branch instead of in the existing one due to the difficulty in directly merging it. This is now fully ready for review! 😃 

Please see #17 for additional history

This closes #17 and closes #12 

It should also close #52 

It also removes prebuilt examples from the dev branch, and adds artifacts to examples from GitHub actions

Requesting review from @formatc1702 and @kvid
closed 2020-07-19 by formatc1702