wireviz/WireViz #13: Apache License?

pdinc-oss, 2020-06-24:
I could get justification from work to contribute, but not on a GPL code base.

Thoughts?

purpleidea, 2020-06-25:
@pdinc-oss Work somewhere else that isn't a hater, or help them get rid of their FUD.

superlou, 2020-06-25:
While @purpleidea's recommendation may not be realistic, what are the cons of GPLv3 for an application like this?

purpleidea, 2020-06-25:
The people who oppose copyleft are the ones it was designed to protect against.

> While @purpleidea's recommendation may not be realistic, what are the cons of GPLv3 for an application like this?

The con is that some company can't use this in their proprietary product.

pdinc-oss, 2020-06-25:
> The people who oppose copyleft are the ones it was designed to protect against.
> 
> > While @purpleidea's recommendation may not be realistic, what are the cons of GPLv3 for an application like this?
> 
> The con is that some company can't use this in their proprietary product.

That is correct, and that is respected. Thank you for the response.

closed 2020-06-25 by pdinc-oss