wireviz/WireViz #14: Support linked part-numbers (Mating connectors, contacts)

aakatz3, 2020-06-24:
It would be nice to add fields within the connectors for mating connectors, and for the crimp contacts for the connectors. While this is not strictly necessary, it would make it easier to specify parts.

kvid, 2020-07-15:
If the main purpose is to include them as BOM entries, then `additional_bom_items` from PR https://github.com/formatc1702/WireViz/pull/80 might be used. If the main purpose is extra information about the connector in the diagram, then the connector attribute `notes` might be used. Issue https://github.com/formatc1702/WireViz/issues/50 assigned to @Tyler-Ward have a more detailed discussion about specifying crimp contacts.

formatc1702, 2020-07-15:
> and for the crimp contacts for the connectors

correct, see #50.

> for mating connectors

I guess this is not BOM relevant (since the mating connector isn't part of the harness itself) but just a hint for the reader, to have an immediate link between matching plug and receptacles... I don't consider it a top priority but I will leave the issue open.