wireviz/WireViz #3: Ability to specify twisted wires

s7726, 2020-06-24:
Add ability to specify, and visualize, twisted wires with appropriate call outs for twist specs.

SnowMB, 2020-10-15:
I made a quick mockup how this could be visually achieved with hacking a little bit on the intermediate files graphviz can produce. This is similar to the solution I proposed in #120 and in my branch https://github.com/SnowMB/WireViz/tree/banded_wires_pydot.

I switched the order of wire row and text row for the 2nd wire in the twisted pairs so I can draw them overlapping without touching the text. I added a few points to the splines that make up the wire inside of the table to produce the twisting effect. This could be tweaked a little bit more to make a better looking.

Actually twisting them (switching which wire is drawn in front for each crossing) is way harder because they have to be split in peaces and each section has to be drawn in a different order.

Graph:
[image]

Hacked file `tutorial02.gv`: 
```
graph G {
outputorder=nodesfirst;
rankdir=LR;
graph [bb="0,0,839,180", bgcolor=white, fontname=arial, nodesep="0.33", rankdir=LR, ranksep=2];
node [fillcolor=white, fontname=arial, label="\N", shape=record, style=filled];
edge [fontname=arial, style=bold];
X1 [height="1.9167", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X1</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Molex KK 254</td><td balign="left">female</td><td balign="left">4-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td port="p1r">1</td></tr><tr><td port="p2r">2</td></tr><tr><td port="p3r">3</td></tr><tr><td port="p4r">4</td></tr></table></td></tr></table>>, margin=0, pos="91,83", shape=none, width="2.5278"];
W1 [height="2.5", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td colspan="3">W1</td></tr><tr><td balign="left">4x</td><td balign="left">0.25 mm² (24 AWG)</td><td balign="left">1 m</td></tr></table></td></tr><tr><td>&nbsp;</td></tr><tr><td><table border="0" cellspacing="0" cellborder="0"><tr><td>X1:1</td><td>WH</td><td>X2:1</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w1" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6"></td></tr><tr><td>X1:2</td><td>BN</td><td>X2:2</td></tr><tr><td>X1:3</td><td>GN</td><td>X2:4</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w3" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w4" height="6"></td></tr><tr><td>X1:4</td><td>YE</td><td>X2:3</td></tr><tr><td>&nbsp;</td></tr></table></td></tr></table>>, margin=0, pos="419.5,90", shape=box, style="", width="2.5972"];
X2 [height="1.9167", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X2</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Molex KK 254</td><td balign="left">female</td><td balign="left">4-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td port="p1l">1</td></tr><tr><td port="p2l">2</td></tr><tr><td port="p3l">3</td></tr><tr><td port="p4l">4</td></tr></table></td></tr></table>>, margin=0, pos="748,85", shape=none, width="2.5278"];
X1:p1r -- X2:p1l  [penwidth="4.0", pos="182.0,95.0 246.01,95.0 261.99,97.0 326.0,97.0 382.1,95.0 400.8,88.0 419.5,88.0 438.2,88.0 475.6,95.0 513.0,97.0 577.0,97.0 593.0,97.0 657.0,97.0", wv_wire=0];
X1:p1r -- X2:p1l  [color="#ffffff", penwidth="3.0", pos="182.0,95.0 246.01,95.0 261.99,97.0 326.0,97.0 382.1,95.0 400.8,88.0 419.5,88.0 438.2,88.0 475.6,95.0 513.0,97.0 577.0,97.0 593.0,97.0 657.0,97.0", wv_wire=0];
X1:p2r -- X2:p2l  [penwidth="4.0", pos="182.0,71.0 246.0,71.0 262.0,88.0 326.0,88.0 382.1,88.0 400.8,95.0 419.5,95.0 438.2,95.0 475.6,88.0 513.0,88.0 577.0,88.0 593.0,73.0 657.0,73.0", wv_wire=1];
X1:p2r -- X2:p2l  [color="#895956", penwidth="3.0", pos="182.0,71.0 246.0,71.0 262.0,88.0 326.0,88.0 382.1,88.0 400.8,95.0 419.5,95.0 438.2,95.0 475.6,88.0 513.0,88.0 577.0,88.0 593.0,73.0 657.0,73.0", wv_wire=1];
X1:p3r -- X2:p4l  [penwidth="4.0", pos="182.0,48.0 246.0,48.0 262.0,47.0 326.0,47.0 382.1,47.0 400.8,38.0 419.5,38.0 438.2,38.0 475.6,47.0 513.0,47.0 577.61,47.0 592.39,27.0 657.0,27.0", wv_wire=2];
X1:p3r -- X2:p4l  [color="#00ff00", penwidth="3.0", pos="182.0,48.0 246.0,48.0 262.0,47.0 326.0,47.0 382.1,47.0 400.8,38.0 419.5,38.0 438.2,38.0 475.6,47.0 513.0,47.0 577.61,47.0 592.39,27.0 657.0,27.0", wv_wire=2];
X1:p4r -- X2:p3l  [penwidth="4.0", pos="182.0,25.0 246.0,25.0 262.0,38.0 326.0,38.0 382.1,38.0 400.8,47.0 419.5,47.0 438.2,47.0 475.6,38.0 513.0,38.0 578.2,38.0 591.8,50.0 657.0,50.0", wv_wire=3];
X1:p4r -- X2:p3l  [color="#ffff00", penwidth="3.0", pos="182.0,25.0 246.0,25.0 262.0,38.0 326.0,38.0 382.1,38.0 400.8,47.0 419.5,47.0 438.2,47.0 475.6,38.0 513.0,38.0 578.2,38.0 591.8,50.0 657.0,50.0", wv_wire=3];
}
```

Run this file with:

```shell
neato -n2 tutorial02.gv -Tpng > test.png
```

formatc1702, 2020-10-16:
@SnowMB It might take some time to get this, or the banded stripes from #120 fully implemented and merged, but I love seeing the results of your GraphViz hackery!

Regarding the definition of the twisted pairs:
My proposal would be to have a new `pairs` attribute within the wire, something like:
```
W1:
  wirecount: 4
  pairs:
    - [1,2]
    - [3,4]
```
or, as was suggested somewhere, even allow adding additional attributes to each pair, such as a twist rate...
```
W1:
  wirecount: 4
  pairs:
    - 
      wires: [1,2]
      twist-rate: 100/m
    - 
      wires: [3,4]
      twist-rate: 95/m
```
You probably wouldn't render the actual difference in twist rates, but it might be a reasonable parameter to include.

SnowMB, 2020-10-20:
> Actually twisting them (switching which wire is drawn in front for each crossing) is way harder because they have to be split in peaces and each section has to be drawn in a different order.

I thought about this a little bit more and actually it's not that hard. The only problem I'm facing right now is that the spliced wire has a noticable black mark where the 2 sections meet.

[image]

@formatc1702 I really like this project. I'm looking forward to the 0.2 release! Maybe after that we can start discussing some fundamental changes that would be needed for my proposal. Feel free to message me when you're ready. Then I'm willing to put in the work in to rebase my branch on the latest changes. So long I'm just using the issues / pr as documentation for my ideas :wink:

formatc1702, 2020-10-20:
Sounds good.
Could you post some renders with 4 and 6 twist crossings instead of 2, just to see how it looks? If it's not too much effort, that is.
Thanks!

SnowMB, 2020-10-20:
Sure. Here you go:

```dot
graph G {
outputorder=nodesfirst;
rankdir=LR;
graph [bb="0,0,1031,280", bgcolor=white, fontname=arial, nodesep="0.33", rankdir=LR, ranksep=2];
node [fillcolor=white, fontname=arial, label="\N", shape=record, style=filled];
edge [fontname=arial, style=bold];
X1 [height="3.1944", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X1</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Stewart Connector SS-37000-002</td><td balign="left">male</td><td balign="left">8-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td>DA+</td><td port="p1r">1</td></tr><tr><td>DA-</td><td port="p2r">2</td></tr><tr><td>DB+</td><td port="p3r">3</td></tr><tr><td>DC+</td><td port="p4r">4</td></tr><tr><td>DC-</td><td port="p5r">5</td></tr><tr><td>DB-</td><td port="p6r">6</td></tr><tr><td>DD+</td><td port="p7r">7</td></tr><tr><td>DD-</td><td port="p8r">8</td></tr></table></td></tr></table>>, margin=0, pos="144.5,133", shape=none, width="4.0139"];
W1 [height="3.8889", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td colspan="4">W1</td></tr><tr><td balign="left">CAT5e</td><td balign="left">8x</td><td balign="left">24 AWG</td><td balign="left">1 m</td></tr></table></td></tr><tr><td>&nbsp;</td></tr><tr><td><table border="0" cellspacing="0" cellborder="0"><tr><td>X1:1</td><td>WHGN</td><td>X2:3</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w1" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6"></td></tr><tr><td>X1:2</td><td>GN</td><td>X2:6</td></tr><tr><td>X1:3</td><td>WHOG</td><td>X2:1</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w3" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w4" height="6"></td></tr><tr><td>X1:4</td><td>OG</td><td>X2:7</td></tr><tr><td>X1:5</td><td>WHBU</td><td>X2:8</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w5" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w6" height="6"></td></tr><tr><td>X1:6</td><td>BU</td><td>X2:2</td></tr><tr><td>X1:7</td><td>WHBN</td><td>X2:4</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w7" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w8" height="6"></td></tr><tr><td>X1:8</td><td>BN</td><td>X2:5</td></tr><tr><td>&nbsp;</td></tr></table></td></tr></table>>, margin=0, pos="515.5,140", shape=box, style="", width="2.2917"];
X2 [height="3.1944", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X2</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Stewart Connector SS-37000-002</td><td balign="left">male</td><td balign="left">8-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td port="p1l">1</td><td>DB+</td></tr><tr><td port="p2l">2</td><td>DB-</td></tr><tr><td port="p3l">3</td><td>DA+</td></tr><tr><td port="p4l">4</td><td>DD+</td></tr><tr><td port="p5l">5</td><td>DD-</td></tr><tr><td port="p6l">6</td><td>DA-</td></tr><tr><td port="p7l">7</td><td>DC+</td></tr><tr><td port="p8l">8</td><td>DC-</td></tr></table></td></tr></table>>, margin=0, pos="886.5,137", shape=none, width="4.0139"];

X1:p2r -- X2:p6l  [penwidth="4.0", pos="289.000,168.000 353.020,168.000 368.980,184.500 433.000,184.500 466.000,184.500 490.750,194.500 515.500,194.500", wv_wire=1];
X1:p2r -- X2:p6l  [color="#00ff00", penwidth="3.0", pos="289.000,168.000 353.020,168.000 368.980,184.500 433.000,184.500 466.000,184.500 490.750,194.500 515.500,194.500", wv_wire=1];
X1:p1r -- X2:p3l  [penwidth="4.0", pos="289.000,191.000 353.060,191.000 368.940,197.000 433.000,197.000 466.000,197.000 490.750,187.000 515.500,187.000 540.250,187.000 565.000,197.000 598.000,197.000 665.460,197.000 674.540,149.000 742.000,149.000", wv_wire=0];
X1:p1r -- X2:p3l  [color="#ffffff", penwidth="3.0", pos="289.000,191.000 353.060,191.000 368.940,197.000 433.000,197.000 466.000,197.000 490.750,187.000 515.500,187.000 540.250,187.000 565.000,197.000 598.000,197.000 665.460,197.000 674.540,149.000 742.000,149.000", wv_wire=0];
X1:p1r -- X2:p3l  [color="#00ff00", penwidth="3.0", pos="289.000,191.000 353.060,191.000 368.940,197.000 433.000,197.000 466.000,197.000 490.750,187.000 515.500,187.000 540.250,187.000 565.000,197.000 598.000,197.000 665.460,197.000 674.540,149.000 742.000,149.000", style=dashed, wv_wire=0];
X1:p2r -- X2:p6l  [penwidth="4.0", pos="515.500,194.500 540.250,194.500 565.000,184.500 598.000,184.500 674.190,184.500 665.810,79.000 742.000,79.000", wv_wire=1];
X1:p2r -- X2:p6l  [color="#00ff00", penwidth="3.0", pos="515.500,194.500 540.250,194.500 565.000,184.500 598.000,184.500 674.190,184.500 665.810,79.000 742.000,79.000", wv_wire=1];

X1:p4r -- X2:p7l  [penwidth="4.0", pos="289.000,121.000 353.000,121.000 369.000,134.500 433.000,134.500 466.000,134.500 490.750,144.500 515.500,144.500", wv_wire=3];
[... cut ...]
X1:p8r -- X2:p5l  [penwidth="4.0", pos="515.500,44.500 540.250,44.500 565.000,34.500 598.000,34.500 671.210,34.500 668.790,102.000 742.000,102.000", wv_wire=7];
X1:p8r -- X2:p5l  [color="#895956", penwidth="3.0", pos="515.500,44.500 540.250,44.500 565.000,34.500 598.000,34.500 671.210,34.500 668.790,102.000 742.000,102.000", wv_wire=7];
}
```

[image]

The wire distances could be tweaked a little bit more, so the image does not look as cluttered, but since I'm doing this by hand for now it's very time consuming to try different parameters. Something to do later when there's an implementation.

formatc1702, 2020-10-20:
Sorry, there was a misunderstanding.
What I meant was, to make the two wires of one pair cross more than two times inside the `W1` node; i.e. white and brown twisted around each other multiple times.

SnowMB, 2020-10-21:
Ah of course. Here ya go

```dot
graph G {
outputorder=nodesfirst;
rankdir=LR;
graph [bb="0,0,1031,280", bgcolor=white, fontname=arial, nodesep="0.33", rankdir=LR, ranksep=2];
node [fillcolor=white, fontname=arial, label="\N", shape=record, style=filled];
edge [fontname=arial, style=bold];
X1 [height="3.1944", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X1</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Stewart Connector SS-37000-002</td><td balign="left">male</td><td balign="left">8-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td>DA+</td><td port="p1r">1</td></tr><tr><td>DA-</td><td port="p2r">2</td></tr><tr><td>DB+</td><td port="p3r">3</td></tr><tr><td>DC+</td><td port="p4r">4</td></tr><tr><td>DC-</td><td port="p5r">5</td></tr><tr><td>DB-</td><td port="p6r">6</td></tr><tr><td>DD+</td><td port="p7r">7</td></tr><tr><td>DD-</td><td port="p8r">8</td></tr></table></td></tr></table>>, margin=0, pos="144.5,133", shape=none, width="4.0139"];
W1 [height="3.8889", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td colspan="4">W1</td></tr><tr><td balign="left">CAT5e</td><td balign="left">8x</td><td balign="left">24 AWG</td><td balign="left">1 m</td></tr></table></td></tr><tr><td>&nbsp;</td></tr><tr><td><table border="0" cellspacing="0" cellborder="0"><tr><td>X1:1</td><td>WHGN</td><td>X2:3</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w1" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6"></td></tr><tr><td>X1:2</td><td>GN</td><td>X2:6</td></tr><tr><td>X1:3</td><td>WHOG</td><td>X2:1</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w3" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w4" height="6"></td></tr><tr><td>X1:4</td><td>OG</td><td>X2:7</td></tr><tr><td>X1:5</td><td>WHBU</td><td>X2:8</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w5" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w6" height="6"></td></tr><tr><td>X1:6</td><td>BU</td><td>X2:2</td></tr><tr><td>X1:7</td><td>WHBN</td><td>X2:4</td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w7" height="6"></td></tr><tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w8" height="6"></td></tr><tr><td>X1:8</td><td>BN</td><td>X2:5</td></tr><tr><td>&nbsp;</td></tr></table></td></tr></table>>, margin=0, pos="515.5,140", shape=box, style="", width="2.2917"];
X2 [height="3.1944", label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">X2</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td balign="left">Stewart Connector SS-37000-002</td><td balign="left">male</td><td balign="left">8-pin</td></tr></table></td></tr><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td port="p1l">1</td><td>DB+</td></tr><tr><td port="p2l">2</td><td>DB-</td></tr><tr><td port="p3l">3</td><td>DA+</td></tr><tr><td port="p4l">4</td><td>DD+</td></tr><tr><td port="p5l">5</td><td>DD-</td></tr><tr><td port="p6l">6</td><td>DA-</td></tr><tr><td port="p7l">7</td><td>DC+</td></tr><tr><td port="p8l">8</td><td>DC-</td></tr></table></td></tr></table>>, margin=0, pos="886.5,137", shape=none, width="4.0139"];

X1:p2r -- X2:p6l  [penwidth="4.0", pos="289.000,191.000 353.060,191.000 368.940,197.000 433.000,197.000 446.750,197.000 460.500,193.000 474.250,193.000", wv_wire=1];
X1:p2r -- X2:p6l  [color="#00ff00", penwidth="3.0", pos="289.000,191.000 353.060,191.000 368.940,197.000 433.000,197.000 446.750,197.000 460.500,193.000 474.250,193.000", wv_wire=1];
X1:p2r -- X2:p6l  [penwidth="4.0", pos="515.500,197.000 529.250,197.000 543.000,193.000 556.750,193.000", wv_wire=1];
X1:p2r -- X2:p6l  [color="#00ff00", penwidth="3.0", pos="515.500,197.000 529.250,197.000 543.000,193.000 556.750,193.000", wv_wire=1];
X1:p1r -- X2:p3l  [penwidth="4.0", pos="289.000,168.000 353.020,168.000 368.980,192.000 433.000,192.000 446.750,192.000 460.500,196.000 474.250,196.000 488.000,196.000 501.750,192.000 515.500,192.000 529.250,192.000 543.000,196.000 556.750,196.000 570.500,196.000 584.250,192.000 598.000,192.000 674.190,192.000 665.810,79.000 742.000,79.000", wv_wire=0];
X1:p1r -- X2:p3l  [color="#ffffff", penwidth="3.0", pos="289.000,168.000 353.020,168.000 368.980,192.000 433.000,192.000 446.750,192.000 460.500,196.000 474.250,196.000 488.000,196.000 501.750,192.000 515.500,192.000 529.250,192.000 543.000,196.000 556.750,196.000 570.500,196.000 584.250,192.000 598.000,192.000 674.190,192.000 665.810,79.000 742.000,79.000", wv_wire=0];
X1:p1r -- X2:p3l  [color="#00ff00", penwidth="3.0", pos="289.000,168.000 353.020,168.000 368.980,192.000 433.000,192.000 446.750,192.000 460.500,196.000 474.250,196.000 488.000,196.000 501.750,192.000 515.500,192.000 529.250,192.000 543.000,196.000 556.750,196.000 570.500,196.000 584.250,192.000 598.000,192.000 674.190,192.000 665.810,79.000 742.000,79.000", style=dashed, wv_wire=0];
[... cut ...]
X1:p8r -- X2:p5l  [color="#895956", penwidth="3.0", pos="474.250,43.000 488.000,43.000 501.750,47.000 515.500,47.000", wv_wire=7];
X1:p8r -- X2:p5l  [penwidth="4.0", pos="556.750,43.000 570.500,43.000 584.250,47.000 598.000,47.000 670.790,47.000 669.210,125.000 742.000,125.000", wv_wire=7];
X1:p8r -- X2:p5l  [color="#895956", penwidth="3.0", pos="556.750,43.000 570.500,43.000 584.250,47.000 598.000,47.000 670.790,47.000 669.210,125.000 742.000,125.000", wv_wire=7];
}
```

[image]

I think this looks pretty neat :smile:

formatc1702, 2020-10-21:
> I think this looks pretty neat 😄

Agree 100% :)

myself248, 2021-05-12:
> Question for the EEs:
> Are there cases where you would twist more than two wires into a bundle?

Yes, there are twisted triads, though in most cases I think the twisting is more for mechanical durability than electrical significance, there might be cases where it matters. (Twisted triad is almost always shielded too, whether it's a solo cable or a component of a larger one.) Also, at least in prototyping, I've seen 3-twist used as an expedient substitute for an individually-shielded-pair with a drain wire. 

(Should individually shielded pairs or subgroups get their own issue?) 

There's also the somewhat special case of star-quad. (Special in that the physical location of the individual conductors is significant, but this can be left to the designer. At some point, there might occur a situation where someone's designing with star-quad but doesn't know what colors it'll be supplied with, and wishes to express "whichever two conductors are diagonally opposite each other". But that should be resolved before anyone's encoding it into YML so I don't think it needs to be provided for.) I think the mechanical details can be ignored here, but support for displaying 4-twist might come in handy.

Personally I would put more-than-2-twist support as a super low priority, but it does exist in the wild.

s7726, 2021-05-12:
There's also this craziness:
https://www.youtube.com/watch?v=GLjKQHMU57k

And this, although calling this a twist might be a stretch.
https://www.rbracing-rsr.com/turbo/wiring_pix/service_loops.jpg

Both of these may very well be their own mechanical thing unnecessary to visualize. And outside the scope of this issue.

myself248, 2021-05-12:
> Both of these may very well be their own mechanical thing unnecessary to visualize. And outside the scope of this issue.

Spot-on. 

IMHO: Twisted-pair support is very important. Individually shielded pair support is very important. Twisted triad and star-quad is not important but still conceivably within scope. 

Rope-lay construction is either part of the cable spec as-ordered, or a mechanical construction detail, and either way is out-of-scope for electrical description. Service loops are a mechanical detail as well. (Unless scope is gonna creep to be a one-stop-shop electrical _and mechanical_ guide, in which case there's a bunch of other stuff that could be useful.) (Again IMHO, those sorts of practices are part of shop workmanship/craft standards, and best described by their own manuals rather than a per-harness diagram.)

jamesfowkes, 2021-08-10:
@SnowMB Is your implementation of twisted wires in your fork?

leoheck, 2021-08-20:
I would like to test it too.

SnowMB, 2021-11-03:
Hey,
I was between jobs and haven't looked at this in a while. Last I looked at this I got stuck with the used graphviz library and played around with using pydot instead. Somewhere in the old issues / prs I documented my progress. (see #120)

The twisted wires was just a quick hack in directly in the intermediate .dot file generated by my branch, so no python implementation available. Sorry 🥲

Knochi, 2022-08-04:
Any progress in this feature? Would be really nice having this for ethernet like cables

rusefillc, 2024-05-05:
Hello from 2024 :)

kvid, 2024-05-05:
@rusefillc wrote:
> Hello from 2024 :)

Welcome to this old discussion about a feature wanted by many. As you can see, @SnowMB did a great job in his PR #120, but it seems he got stuck with limitations in "the used graphviz library". His contributions here were based on his code in that PR, but with time-consuming extra manual editing of the `.gv` file to twist the wire splines.

If you or someone else are motivated to dive into such low-level Graphviz hacking and (even better) create code to automate it, your contributions into this project are more than welcome!

martinrieder, 2024-05-13:
You can achieve simply this through flipping the order of wires in a bundle. You could also repeat that multiple times and connect these with "virtual splices". I think that putting a comment beneath the wire description is by far more explicit than any layout trickery:
[image]

kvid, 2024-05-13:
@martinrieder wrote:
> You can achieve simply this through flipping the order of wires in a bundle.

That would only make a twist between the bundle box and the connector box. 

If you look at the graphic examples earlier in this issue, you can see that twisted wires _inside_ the cable box is the wanted feature.

> You could also repeat that multiple times and connect these with "virtual splices".

That might work, but be aware that such virtual splices are implemented as nodes in Graphviz, and will demand rank space as other nodes in the diagram. Try it out, to see the demanded space for each twist.

> I think that putting a comment beneath the wire description is by far more explicit than any layout trickery: 

In the current version, you can add `notes` common for the whole cable/bundle.

martinrieder, 2024-05-14:
Thanks for pointing out the differences to the approach above by @SnowMB.  The issue description by @s7726 leaves it open whether the twists should be shown inside or outside the wire box. It is of course more accurate to depict the twisting along with the wire definitions inside the cable box.

In simple cases, it might be sufficient to have Graphviz do the twisting outside and just hide the cable and connector boxes using a tweak. 
[image]

```
templates:
  - &CONN
    pincount: 2
    show_name: false
    show_pincount: false
  - &WIRE
    wirecount: 2
    show_name: false
    show_wirecount: false
    show_wirenumbers: false

connectors:
  Conn0:
    pincount: 2
  Conn1:
    <<: *CONN
  Conn2:
    <<: *CONN
  Conn3:
    <<: *CONN
    
cables:
  Wire1:
    <<: *WIRE
  Wire2:
    <<: *WIRE
    show_wirenumbers: true
  Wire3:
    <<: *WIRE

connections:
  -
    - Conn0: [1,2]
    - Wire1: [2,1]
    - Conn1: [1,2]
    - Wire2: [2,1]
    - Conn2: [1,2]
    - Wire3: [2,1]
    - Conn3: [1,2]

tweak:
    override:
        Wire1:
            shape: plain 
        Wire2:
            shape: none
        Wire3:
            shape: none
            style: invis
            fixedsize: "true"
        Conn1:
            shape: none
            style: invis
            fixedsize: "true"
        Conn2:
            shape: none
            style: invis
            fixedsize: "true"
        Conn3:
            shape: plain

```

Note the difference between styles `none` and `plain`, whereof the latter supports HTML, just without the surrounding `box` style.
The real trick here is to set `fixedsize` to `"true"` (as a string), which apparently sets the `width` attribute to its minimum value of 0.01.

EldarPolitkin, 2025-10-16:
[IEC_60617_SYMBOLS.pdf](https://github.com/user-attachments/files/22945487/IEC_60617_SYMBOLS.pdf)
IEC 60617 

this is what i know from my history and couple of companies and tools that i used - like Altium