wireviz/WireViz #6: Ability to add internal part numbers and special instructions

sahil48, 2020-06-24:
It would be great to be able to add our internal part numbers for different wires and connectors and notes with special instructions (ex. how much to wind the wires, how much insulation to cut, warnings, etc.

kvid, 2020-06-28:
You can already put special instructions in the connector and cable `notes` attribute. See how in the https://github.com/formatc1702/WireViz/tree/master/tutorial number 4, and if https://github.com/formatc1702/WireViz/pull/28 is accepted, then you can put internal part numbers as leading or trailing part of the connector and cable `type` attribute. Would these two attributes solve your issue?

formatc1702, 2020-06-28:
Discussion about part numbers is already happening in #11, other instructions can be placed in the `notes` section as @kvid has mentioned.

closed 2020-06-28 by formatc1702