wireviz/WireViz #19: Add scripting support

ceremcem, 2020-06-25:
Hi, 

It always ended up with trying to turn the configurations into a script in my cases. I would like to know if you'd like to have scripting support for configuration files?

My proposal is very straightforward: Using LiveScript as a "configuration format" will do everything we need: 

* Reusability 
* Modularity 
* Real inheritance support
* Scriptability 

Here is an example LiveScript configuration that looks very clean like YAML: https://gist.github.com/ceremcem/4ca8d774e660c579682839620d8c2da8

formatc1702, 2020-06-28:
Hi, could you provide a specific example of how you would use this for WireViz? I am not sure I understand what it is you want to turn into a script... if this is a separate add-on and not a feature for WireViz specifically, as you mention in your edit, I would like to close this issue. Thanks!

ceremcem, 2020-06-30:
This suggestion popped out of this need of mine: https://stackoverflow.com/questions/14184971/more-complex-inheritance-in-yaml

The very first usage might be "component based design". Some sort of wires can be connected to some sort of connectors and this preprepared setups can be used in another project; which my suggestion makes it possible. 

This "feature" can be handed over a pre-process, but this requires WirViz to accept pure JSON files as well as the YAML files.