wireviz/WireViz #21: First example doesn't match

purpleidea, 2020-06-25:
The connections don't match the image, or I'm misunderstanding the triplets of integers.

Thanks!

formatc1702, 2020-06-27:
If you're referring to [Demo 01](https://github.com/formatc1702/WireViz#demo-01):

`  -
    - X1: [5,2,1]
    - W1: [1,2,3]
    - X2: [1,3,2]`

You need to read the columns from top to bottom:
Pin 5 of X1 goes through wire 1 of W1 into pin 1 on X2.
Pin 2 of X1 goes through wire 2 of W1 into pin 3 on X2.
Pin 1 of X1 goes through wire 3 of W1 into pin 2 on X2.

If you're referring to [Example 01](https://github.com/formatc1702/WireViz/tree/master/examples#example-01):

`[1-4]` expands to `[1,2,3,4]`... from there, see explanation above.

closed 2020-06-27 by formatc1702

purpleidea, 2020-06-27:
@formatc1702 

> You need to read the columns from top to bottom:

That explains it. From a programming perspective, this is very confusing, is there a reason each wire is split *across* the yaml area instead of a wire being defined *within* an array?

At the very least there should be a good comment explaining this matrix transpose.

Thanks!

formatc1702, 2020-06-27:
I am open for suggestions of how this could be implemented more cleanly! :)

The current syntax makes it very easy to define straight one-to-one connections ([Tutorial 01](https://github.com/formatc1702/WireViz/blob/master/tutorial/readme.md#4---templates-ii-notes-american-standards-daisy-chaining-i)) and auto-generating ferrules that don't have a visible designator ([Tutorial 05](https://github.com/formatc1702/WireViz/blob/master/tutorial/readme.md#5---ferrules-wire-bundles-custom-wire-colors)).

If there's a more elegant way to define this, I am up for changing things around.

purpleidea, 2020-06-27:
@formatc1702 This is not my speciality, but in my subjective brain I would think one array would be one wire. In other words the transpose of what you have now. https://en.wikipedia.org/wiki/Transpose