wireviz/WireViz #22: ℃ cannot be printed

tszabooatIMT, 2020-06-26:
Defining working temperature is not possible. It is printed as either â„ƒ or Â°C depending on which character is used. Looks like an UTF coding problem. Maybe it could be defined in an other way, like parameter?

kvid, 2020-06-26:
Could you provide a YAML input file and an output file that shows this problem? I tried to append `40°C` to the connector type in ex05.yml and my case all output files showed the temperature as expected without any extra characters.
`    type: Molex KK 254 40°C`
[image]

tszabooatIMT, 2020-06-26:
[image]

connectors:
  X1:
    type: example2
    subtype: RA
    pinout: [A,B,C,D]
  X2:
    type: example1
    subtype: RA
    pinout: [VP, Rxd/TxD-N, DGND, Rxd/TxD-N]

cables:
  W1:
    gauge: 0.25 mm2
    length: 1
    colors: [RD, WH, BK, GN]
    notes: Working temperature -40°C  +60°C (fixed installation), 500Vrsm AC voltage rating
connections:
  -
    - X1: [1,2,3,4]
    - W1: [1,2,3,4]
    - X2: [1,2,3,4]

kvid, 2020-06-26:
Your YAML input did not survive completely in your last posting. Try the "Insert code" button (look for <> in the toolbar) and check the result with "Preview" when you need to keep spacing/indentation, etc. However, I tried to reconstruct it into valid syntax (please verify my reconstruction below against your input file): 
```
connectors:
  X1:
    type: example2
    subtype: RA
    pinout: [A,B,C,D]
  X2:
    type: example1
    subtype: RA
    pinout: [VP, Rxd/TxD-N, DGND, Rxd/TxD-N]

cables:
  W1:
    gauge: 0.25 mm2
    length: 1
    colors: [RD, WH, BK, GN]
    notes: Working temperature -40°C +60°C (fixed installation), 500Vrsm AC voltage rating

connections:
  -
    - X1: [1,2,3,4]
    - W1: [1,2,3,4]
    - X2: [1,2,3,4]
```
With the input above, the output is still as expected without any extra characters in my case: 
[image]
Can you verify that we use the same software version? I execute Python from Ubuntu bash installed as [WSL 1 in Win10](https://docs.microsoft.com/en-us/windows/wsl/install-win10), just in case that might make a difference.
```
kvid@PC:~/src/formatc1702/WireViz/src$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic
kvid@PC:~/src/formatc1702/WireViz/src$ python3.7 --version
Python 3.7.5
kvid@PC:~/src/formatc1702/WireViz/src$ git remote -v
origin  https://github.com/formatc1702/WireViz.git (fetch)
origin  https://github.com/formatc1702/WireViz.git (push)
kvid@PC:~/src/formatc1702/WireViz/src$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
kvid@PC:~/src/formatc1702/WireViz/src$ git log HEAD~1..HEAD
commit 66a47169e621d9605d783aabe65a0097e28609c2 (HEAD -> master, origin/master, origin/HEAD)
Merge: 6406e28 c1e6cf1
Author: Daniel Rojas <github@danielrojas.net>
Date:   Thu Jun 18 12:55:20 2020 +0200

    Merge branch 'dev'

commit c1e6cf14d2ea93f65d5c14ae8e3f30babcc4d8f9 (origin/dev)
Author: Daniel Rojas <github@danielrojas.net>
Date:   Thu Jun 18 12:55:04 2020 +0200

    Update readme
```

kvid, 2020-06-26:
Please also verify that you get this output/intermediate file (file extension .gv) that is the input file for Graphviz: 
```
graph {
// Graph generated by WireViz
// https://github.com/formatc1702/WireViz
	graph [bgcolor=white fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
	node [fillcolor=white fontname=arial shape=record style=filled]
	edge [fontname=arial style=bold]
	X1 [label="X1|{example2|RA|4-pin}|{{A|B|C|D}|{<p1r>1|<p2r>2|<p3r>3|<p4r>4}}"]
	X2 [label="X2|{example1|RA|4-pin}|{{<p1l>1|<p2l>2|<p3l>3|<p4l>4}|{VP|Rxd/TxD-N|DGND|Rxd/TxD-N}}"]
	edge [color="#000000:#ff0000:#000000"]
	X1:p1r:e -- W1:w1:w
	W1:w1:e -- X2:p1l:w
	edge [color="#000000:#ffffff:#000000"]
	X1:p2r:e -- W1:w2:w
	W1:w2:e -- X2:p2l:w
	edge [color="#000000:#000000:#000000"]
	X1:p3r:e -- W1:w3:w
	W1:w3:e -- X2:p3l:w
	edge [color="#000000:#00ff00:#000000"]
	X1:p4r:e -- W1:w4:w
	W1:w4:e -- X2:p4l:w
	W1 [label=<<table border="0" cellspacing="0" cellpadding="0"><tr><td><table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr><td colspan="3">W1</td></tr><tr><td>4x</td><td>0.25 mm²</td><td>1 m</td></tr></table></td></tr><tr><td>&nbsp;</td></tr><tr><td><table border="0" cellspacing="0" cellborder="0"><tr><td>X1:1</td><td>RD</td><td>X2:1</td></tr><tr><td colspan="3" cellpadding="0" height="6" bgcolor="#ff0000" border="2" sides="tb" port="w1"></td></tr><tr><td>X1:2</td><td>WH</td><td>X2:2</td></tr><tr><td colspan="3" cellpadding="0" height="6" bgcolor="#ffffff" border="2" sides="tb" port="w2"></td></tr><tr><td>X1:3</td><td>BK</td><td>X2:3</td></tr><tr><td colspan="3" cellpadding="0" height="6" bgcolor="#000000" border="2" sides="tb" port="w3"></td></tr><tr><td>X1:4</td><td>GN</td><td>X2:4</td></tr><tr><td colspan="3" cellpadding="0" height="6" bgcolor="#00ff00" border="2" sides="tb" port="w4"></td></tr><tr><td>&nbsp;</td></tr></table></td></tr><tr><td cellpadding="3">Working temperature -40°C +60°C (fixed installation), 500Vrsm AC voltage rating</td></tr><tr><td>&nbsp;</td></tr></table>> fillcolor=white margin=0 shape=box style=""]
}
```
If I open the file above in a [text viewer I often use](https://www.ghisler.com/), it automatically detect it to contain UTF-8 and show the text correctly, but if I force it to show the file using ANSI (Windows charset), then I can see these extra characters: `Working temperature -40Â°C +60Â°C (fixed installation), 500Vrsm AC voltage rating` inside the file above. If your .gv file is identical to mine as shown above assuming UTF-8 encoding, then I agree with you that it might look like your problem might be caused by your Graphviz version assumes the file above to be ANSI encoded instead of UTF-8. Please check the character encoding of the other output files as well. The default character encoding might be OS and browser dependent and can also be changed by the user, but my experience is that modern software today normally is able to autodetect this correctly with default settings. If this is your problem, then all non-ASCII characters should be affected. Here are the raw bytes of the section of the .gv file containing non-ASCII characters: 
[image]

tszabooatIMT, 2020-06-26:
The YML file you posted is identical.
This is in my .gv file (renamed to txt, as git doesnt support gv files)
[example.txt](https://github.com/formatc1702/WireViz/files/4838906/example.txt)

 "-40Â°C +60Â°C "
OS is windows 10

kvid, 2020-06-30:
I'm sorry, but I am currently unable to test this in Win10. Maybe someone else can test your yaml file in Win10. Have you tried the latest release? A lot of changes was committed the last few days.

formatc1702, 2020-07-05:
I'm closing this but keeping a reference in #52 since it seems to be a specific instance of the broader problem described there. Since I don't use Windows, I can't offer any help on this.

closed 2020-07-05 by formatc1702