wireviz/WireViz #23: How to execute?

martingcavallo, 2020-06-26:
Hi,

Could anybody explain me how too execute this tool in windows?

I understand that it executes from command line, in the wireviz.py directory (C:\WireViz\src in my case), running "python3 wireviz.py filename.yml" command.

is this right? I ran this line but nothing habppened.

Thanks

pandorazboxx, 2020-06-26:
From the readme:

> ## Usage
>
>```
>$ python3 wireviz.py ~/path/to/file/mywire.yml
>```
>
>This will output the following files
>
>```
>mywire.gv         GraphViz output
>mywire.svg        Wiring diagram as vector image
>mywire.png        Wiring diagram as raster image
>mywire.bom.tsv    BOM (bill of materials) as tab-separated text file
>mywire.html       HTML page with wiring diagram and BOM embedded
>```

If you are using git bash for a terminal, you can simply type `ls` to see if those files are there now.

In Windows explorer you can just navigate to that folder to see those files.

I'm assuming your yml file follows the correct format?

cookieSN, 2020-06-27:
Hi,

I have a problem to execute my test file. I'm not an expert in python, so following I had done:

1. Download and install latest Python Version for Windows (3.8.3)
2. Set Path Variables for Python -> in my case ("c:users/$yourName$/appdata/local/programs/pyhton/Python38-32
3. Open Git Bash with admin right and go to the folder, where the src folder of wireviz is
4. execute by following comand: python wireviz.py first_test.yml

In the file "first test.yml" is the code from "ex01.yml". Unfortunately, there were a few error messages while executing. 

Do anyone know what I did wrong?

[image]

formatc1702, 2020-06-28:
Please get the latest version of WireViz and make sure you follow the [installation guide](https://github.com/formatc1702/WireViz#installation), since it seems you don't have GraphViz (not the Python module, but the [actual executable](https://graphviz.org/download/)) installed.

If this works, please close the issue. Good luck!

slightlynybbled, 2020-07-10:
Seems like this one should be closed at this time.

closed 2020-07-10 by formatc1702