wireviz/WireViz #26: Add support for cable splicing

formatc1702, 2020-06-27:
Possibly using a "pseudo-connector" linking e.g. one wire from the left and two wires from the right.

Proof-of-concept:

```
connectors:
  X1:
    pincount: 5
  X2:
    pincount: 8
  S1:
    category: ferrule # TODO: splice
    pincount: 1
    type: Solderjoint + Tape
    notes: splice
  S2:
    category: ferrule # TODO: splice
    pincount: 1
    type: Solderjoint + Tape
    notes: splice

cables:
  W1: &template_wire
    category: bundle
    wirecount: 3
    gauge: 0.25
    length: 1
    show_name: false
    colors: [PK]
  W2:
    <<: *template_wire
    wirecount: 2
    length: 0.8
    colors: [TQ, YE]
  W3:
    <<: *template_wire
    wirecount: 5
    length: 0.2
    colors: [TQ, TQ, YE, YE, YE]

connections:
  -
    - X1: [1-3]
    - W1: [1-3]
    - X2: [1-3]
  -
    - X1: [4,5]
    - W2: [1,2]
  -
    - W2: 1
    - S1: 1
  -
    - W2: 2
    - S2: 1
  -
    - S1: [1,1]
    - W3: [1,2]
    - X2: [4,5]
  -
    - S2: [1,1,1]
    - W3: [3,4,5]
    - X2: [6,7,8]
```

[image]

kvid, 2020-06-27:
Really nice! I guess a final solution can avoid the duplicated text "Solderjoint + Tape" inside the splice frame. :-)

formatc1702, 2020-06-27:
Yes, this is just an artifact of the way ferrules are implemented now, since they initially were designed to have only one pin, and one wire going into it... splicing is on the roadmap, though!

I'm not quite sure why GraphViz makes the topmost wire bend like that, though... that's something I have little control over...

formatc1702, 2020-07-05:
I refactores the entire ferrule/splice section, the bug with the double label is fixed, and splices now look good regardless of whether few or many attributes are specified.

[image]

I am considering to change the `ferrule` category name to `simple`, to encompass both ferrules, splices, and any other simple connectors that don't have a designator and essentially only have one port (electrically speaking, at least).

kvid, 2020-07-06:
I suggest this also can be used for any product to connect several wires in one physical point, e.g. a Wire Nut, as mentioned in issue https://github.com/formatc1702/WireViz/issues/45#issuecomment-654097778

formatc1702, 2020-07-06:
I consider this completed and have included some of the thoughts in #67.

closed 2020-07-06 by formatc1702