wireviz/WireViz #29: Visualize wire gauge differences as differences in wire thickness

kvid, 2020-06-28:
If we go in this direction, then maybe a difference in cable gauge also optionally could be visualized as a difference in wire thickness between different cables. Equal wire thickness can be implemented using the same number of color entries for these wires internally and for Graphviz, i.e. several equal color entries for unicolored wires.

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/17#issuecomment-650377638_

kvid, 2020-06-28:
AFAIK, the wire gauge value is not currently inerpreted, but one approach could be to sort all unique gauge values, use the minimum thickness for the smallest mm^2 value (or highest AWG), add a constant to the thickness for the next gauge value, etc. If we need a maximum thicness limit, then close gauge values can be grouped and use equal thickness. Alternatively, a less functional, but simpler implementation could be to add an optional wire thickness attribute in the YAML file.

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/17#issuecomment-650544445_

kvid, 2020-07-06:
I can try to do this after https://github.com/formatc1702/WireViz/pull/17 is merged in.

formatc1702, 2020-07-06:
Much appreciated. Let's hope the merge can happen soon.