wireviz/WireViz #38: Port unrecognized when specifying connections using pinout

Marsfan, 2020-06-28:
### Summary
I am trying to set up connections using the pin names specified in pinouts like in #10. It appears to be working properly, however, when I build the diagram, I get the warning. 

`Warning: node x1, port pGNDr unrecognized`

I am not sure what is happening, could you clear this up for me?

### Minimal Working Example
#### YAML File

```
connectors:
  X2:
    pinnumbers: [1,2]
    pinout: [GND, VCC]
  X1:
    pinnumbers: [1, 2, 3, 4]
    pinout: [GND, VCC, D+, D-]

cables:
  W1:
    wirecount: 2
    color_code: IEC

connections:
  -
    - X1: [GND, VCC]
    - W1: [1-2]
    - X2: [1-2]
```

#### Command Line Input/Output
```
$> wireviz test.yml
Warning: node X1, port pGNDr unrecognized
Warning: node X1, port pVCCr unrecognized
Warning: node X1, port pGNDr unrecognized
Warning: node X1, port pVCCr unrecognized
$>
```

#### Output from running on provided YAML file
[image]

formatc1702, 2020-06-29:
I think there's a misunderstanding. You need to use items from the `pinnumbers` list in your `connections`, not from `pinouts`. The reason is, `pinnumbers` are unambiguous, whereas your `pinout` might include, for example, several `GND` pins; which one should the connection go to? (see [this example from the tutorial](https://github.com/formatc1702/WireViz/blob/dev/tutorial/readme.md#5---ferrules-wire-bundles-custom-wire-colors))

You _can_ make your `pinnumbers` non-numerical if you want (see X2 in #10), but this is separate from specifying the pin _function_. Hope it helps!

closed 2020-06-29 by formatc1702

Marsfan, 2020-06-29:
Ahh. That makes sense. Thank you very much.

formatc1702, 2020-07-09:
@Marsfan you'll be happy to know that the example you provided above now works in the latest version of the dev branch after closing #71. You can now refer to pin names from `pinout` instead of `pinnumbers`, provided they are unambiguous. Unconnected pins are not checked for ambiguity, which allows to have, for example, multiple pins named `N/C`.