wireviz/WireViz #43: How to install correctly after refactoring?

kvid, 2020-06-29:
Yesterday morning, I was running WireViz (latest release at that time) and created a small pull request with a proposed feature enhancement. Since then, a major refactoring was done, and I now need to use the latest release to update my pull request. That was not so easy as I hoped.

1. First, I tried using my existing repo, checking out master branch and pulling the newest release from this public repo. That went OK.

1. I realized the embedded Graphviz classes was gone, and tried `pip3 install -e .` from the main folder as described in the [README](https://github.com/formatc1702/WireViz/blob/master/README.md), but still I got `ModuleNotFoundError: No module named 'graphviz'`.

1. I tried `sudo apt-get install python-apt` as suggested at [stackoverflow](https://stackoverflow.com/questions/13708180/python-dev-installation-error-importerror-no-module-named-apt-pkg) since I also got the `ModuleNotFoundError: No module named 'apt_pkg'` error, but still no luck.

1. I tried `sudo apt-get install graphviz libgraphviz-dev pkg-config` as I found in a [check log for a recent pull request](https://github.com/formatc1702/WireViz/pull/39/checks), but still no luck.

1. I tried `pip3 install graphviz` as described [here](https://pypi.org/project/graphviz/), but still no luck.

1. I tried creating a new clean local repo, and install again, but still no luck:
```
git clone https://github.com/formatc1702/WireViz.git
cd WireViz/
pip3 install -e .
cd src/wireviz
python3.7 build_examples.py
```
Here is the full Traceback:
```
Traceback (most recent call last):
  File "build_examples.py", line 8, in <module>
    from wireviz import wireviz
  File "../wireviz/wireviz.py", line 6, in <module>
    from graphviz import Graph
ModuleNotFoundError: No module named 'graphviz'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 24, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'

Original exception was:
Traceback (most recent call last):
  File "build_examples.py", line 8, in <module>
    from wireviz import wireviz
  File "../wireviz/wireviz.py", line 6, in <module>
    from graphviz import Graph
ModuleNotFoundError: No module named 'graphviz'
```

1. My running environment is described in https://github.com/formatc1702/WireViz/issues/22#issuecomment-650238895 and just ask if you need more information. Any help on what I should try is appreciated.

This is my fifth trial to create this issue. Github is slow and not very stable today.

formatc1702, 2020-06-29:
The following workflow works for me. I'm on a Mac, it might differ slightly on Linux.

```
$ git clone https://github.com/formatc1702/WireViz.git
$ cd WireViz/
$ python -m venv wv-env
$ source wv-env/bin/activate
$ pip3 install -e .
$ cd src/wireviz
$ python build_examples.py
```

Unfortunately, I'm not an expert, so if the above does not help you, perhaps someone else can chime in?

kvid, 2020-06-29:
The `python3-venv` installation seems to have been completed, but still i am asked to install it again. I have not used this package before, but if if might help me. I'm willing to try. I'm also asked to `recreate your virtual environment`. Is that done with your suggested command `python -m venv wv-env`, or do I have to run a different command first?
```
kvid@PC:~/src/formatc1702$ rm -rf WireViz/
kvid@PC:~/src/formatc1702$ git clone https://github.com/formatc1702/WireViz.git
Cloning into 'WireViz'...
remote: Enumerating objects: 164, done.
remote: Counting objects: 100% (164/164), done.
remote: Compressing objects: 100% (97/97), done.
remote: Total 1183 (delta 81), reused 123 (delta 52), pack-reused 1019
Receiving objects: 100% (1183/1183), 5.05 MiB | 555.00 KiB/s, done.
Resolving deltas: 100% (737/737), done.
kvid@PC:~/src/formatc1702$ cd WireViz/
kvid@PC:~/src/formatc1702/WireViz$ python3.7 -m venv wv-env
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/kvid/src/formatc1702/WireViz/wv-env/bin/python3.7', '-Im', 'ensurepip', '--upgrade', '--default-pip']

kvid@PC:~/src/formatc1702/WireViz$ sudo apt-get install python3-venv
[sudo] password for kvid:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libpython3.6 libpython3.6-dev libpython3.6-minimal libpython3.6-stdlib python3.6 python3.6-dev python3.6-minimal
  python3.6-venv
Suggested packages:
  python3.6-doc binfmt-support
The following NEW packages will be installed:
  python3-venv python3.6-venv
The following packages will be upgraded:
  libpython3.6 libpython3.6-dev libpython3.6-minimal libpython3.6-stdlib python3.6 python3.6-dev python3.6-minimal
7 upgraded, 2 newly installed, 0 to remove and 134 not upgraded.
Need to get 50.9 MB of archives.
After this operation, 100 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3.6-dev amd64 3.6.9-1~18.04ubuntu1 [508 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 libpython3.6-dev amd64 3.6.9-1~18.04ubuntu1 [44.9 MB]
Get:3 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 libpython3.6 amd64 3.6.9-1~18.04ubuntu1 [1414 kB]
Get:4 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3.6 amd64 3.6.9-1~18.04ubuntu1 [203 kB]
Get:5 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 libpython3.6-stdlib amd64 3.6.9-1~18.04ubuntu1 [1710 kB]
Get:6 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3.6-minimal amd64 3.6.9-1~18.04ubuntu1 [1609 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 libpython3.6-minimal amd64 3.6.9-1~18.04ubuntu1 [533 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 python3.6-venv amd64 3.6.9-1~18.04ubuntu1 [6188 B]
Get:9 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 python3-venv amd64 3.6.7-1~18.04 [1208 B]
Fetched 50.9 MB in 5s (10.6 MB/s)
(Reading database ... 60676 files and directories currently installed.)
Preparing to unpack .../0-python3.6-dev_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking python3.6-dev (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../1-libpython3.6-dev_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking libpython3.6-dev:amd64 (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../2-libpython3.6_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking libpython3.6:amd64 (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../3-python3.6_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking python3.6 (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../4-libpython3.6-stdlib_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking libpython3.6-stdlib:amd64 (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../5-python3.6-minimal_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking python3.6-minimal (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Preparing to unpack .../6-libpython3.6-minimal_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking libpython3.6-minimal:amd64 (3.6.9-1~18.04ubuntu1) over (3.6.9-1~18.04) ...
Selecting previously unselected package python3.6-venv.
Preparing to unpack .../7-python3.6-venv_3.6.9-1~18.04ubuntu1_amd64.deb ...
Unpacking python3.6-venv (3.6.9-1~18.04ubuntu1) ...
[... 22 lines cut ...]
You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/kvid/src/formatc1702/WireViz/wv-env/bin/python3.7', '-Im', 'ensurepip', '--upgrade', '--default-pip']

kvid@PC:~/src/formatc1702/WireViz$ sudo apt-get install python3-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-venv is already the newest version (3.6.7-1~18.04).
0 upgraded, 0 newly installed, 0 to remove and 134 not upgraded.
kvid@PC:~/src/formatc1702/WireViz$ python3.7 -m venv wv-env
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/kvid/src/formatc1702/WireViz/wv-env/bin/python3.7', '-Im', 'ensurepip', '--upgrade', '--default-pip']

kvid@PC:~/src/formatc1702/WireViz$
```

formatc1702, 2020-06-29:
Perhaps [this StackOverflow thread](https://stackoverflow.com/questions/39539110/pyvenv-not-working-because-ensurepip-is-not-available) might help, someone mentions WLS as well, which you mentioned you are using... other than that, I'm afraid I can't offer any valuable suggestions.. sorry! Hope you get it working again soon!

Marsfan, 2020-06-29:
@kvid I am the person who made the refactoring. My apologies for messing up your system. After I finished the refactoring, this is what I did to bring everything up to date as an install. 

``` 
pip3 uninstall wireviz
pip3 install -e .
```

As for re-creating the virtual environment, did you delete the existing environment before running `python -m venv wv-env`? 

I did not need to install any 3rd party packages after the refactoring (other than those that were already installed).

kvid, 2020-06-29:
@formatc1702 I tried the commands setting the locale from the page you linked to, with no luck, but one of the comments mentioned an alternative command that worked for him, and by changing the version number to 3.7 as I use, I tried this command:
`sudo apt-get install python3.7-venv`
That seemed to help:
```
kvid@PC:~/src/formatc1702/WireViz$ sudo apt-get install python3.7-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  python3.7-venv
0 upgraded, 1 newly installed, 0 to remove and 134 not upgraded.
Need to get 6184 B of archives.
After this operation, 32.8 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 python3.7-venv amd64 3.7.5-2~18.04 [6184 B]
Fetched 6184 B in 0s (50.2 kB/s)
[master 073cfb3] saving uncommitted changes in /etc prior to apt run
 1 file changed, 1 insertion(+), 1 deletion(-)
Selecting previously unselected package python3.7-venv.
(Reading database ... 60687 files and directories currently installed.)
Preparing to unpack .../python3.7-venv_3.7.5-2~18.04_amd64.deb ...
Unpacking python3.7-venv (3.7.5-2~18.04) ...
Setting up python3.7-venv (3.7.5-2~18.04) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
kvid@PC:~/src/formatc1702/WireViz$ ls
LICENSE  README.md  examples  requirements.txt  setup.py  src  tutorial  wv-env
kvid@PC:~/src/formatc1702/WireViz$ python3.7 -m venv wv-env
kvid@PC:~/src/formatc1702/WireViz$ source wv-env/bin/activate
(wv-env) kvid@PC:~/src/formatc1702/WireViz$ pip3 install -e .
Obtaining file:///home/kvid/src/formatc1702/WireViz
Collecting graphviz (from wireviz==0.1)
  Using cached https://files.pythonhosted.org/packages/83/cc/c62100906d30f95d46451c15eb407da7db201e30f42008f3643945910373/graphviz-0.14-py2.py3-none-any.whl
Collecting pyyaml (from wireviz==0.1)
  Using cached https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f9501676c988a5411ed5572f35d1beff9cae07d321512c/PyYAML-5.3.1.tar.gz
Building wheels for collected packages: pyyaml
  Running setup.py bdist_wheel for pyyaml ... error
  Complete output from command /home/kvid/src/formatc1702/WireViz/wv-env/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-t90e4qx3/pyyaml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpb7ogb1fdpip-wheel- --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for pyyaml
  Running setup.py clean for pyyaml
Failed to build pyyaml
Installing collected packages: graphviz, pyyaml, wireviz
  Running setup.py install for pyyaml ... done
  Running setup.py develop for wireviz
Successfully installed graphviz-0.14 pyyaml-5.3.1 wireviz
(wv-env) kvid@PC:~/src/formatc1702/WireViz$ cd src/wireviz
(wv-env) kvid@PC:~/src/formatc1702/WireViz/src/wireviz$ python build_examples.py
../../examples/demo01.yml
../../examples/demo02.yml
../../examples/ex01.yml
../../examples/ex02.yml
../../examples/ex03.yml
../../examples/ex04.yml
../../examples/ex05.yml
../../examples/ex06.yml
../../examples/ex07.yml
../../examples/ex08.yml
../../examples/ex09.yml
../../tutorial/tutorial01.yml
../../tutorial/tutorial02.yml
../../tutorial/tutorial03.yml
../../tutorial/tutorial04.yml
../../tutorial/tutorial05.yml
../../tutorial/tutorial06.yml
[... 54 lines cut ...]
        modified:   ../../tutorial/tutorial05.html
        modified:   ../../tutorial/tutorial05.png
        modified:   ../../tutorial/tutorial05.svg
        modified:   ../../tutorial/tutorial06.html
        modified:   ../../tutorial/tutorial06.png
        modified:   ../../tutorial/tutorial06.svg
        modified:   ../../tutorial/tutorial07.html
        modified:   ../../tutorial/tutorial07.png
        modified:   ../../tutorial/tutorial07.svg

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        ../../wv-env/

no changes added to commit (use "git add" and/or "git commit -a")
(wv-env) kvid@PC:~/src/formatc1702/WireViz/src/wireviz$
```
I did get the error `Failed building wheel for pyyaml`, but building WireViz examples seemed to work anyway. Now, I can try this with my own repo, but the next few days, I'll be on a family vacation, and might not have much time for this. Thank you for your effort helping me!

PS: When I did a `git diff`, I saw the modified files have quite a lot of differences, but I guess this one explains that:
```
-<!-- Generated by graphviz version 2.44.0 (20200408.0750)
+<!-- Generated by graphviz version 2.40.1 (20161225.0304)
```

formatc1702, 2020-07-04:
Did you get it working without any errors? Feel free to close the issue anytime; if you still need help, I'll leave it open.

kvid, 2020-07-04:
I don't know why I got the error `Failed building wheel for pyyaml`, as shown above, but I have not seen any problems lately. However, I have not had time to try cloning a fresh repo and creating virtual environment again, or even better, trying on a different machine or clean docker image - maybe later this summer, after my vacation, but for now, I don't need more help about this. I therefore close this issue. Thanks for all the help!

closed 2020-07-04 by kvid