wireviz/WireViz #45: [feature] Support for Wago connectors

gioele, 2020-06-30:
Wago connectors are a common kind of splicing connectors, a modern alternative to twist-on wire connectors.

[image]

Differently from in-line splicing connectors (see feature request #26), wires in Wago connectors are all on the same side of the connector. It would be nice if this important property could be carried over in the graphical representation.

aakatz3, 2020-06-30:
To add onto this, Wire Nuts (common in USA installations) behave relatively similarly, and can likely be implemented at the same time.

[image]

kvid, 2020-07-06:
The wire nut have all the wires connected in the same "hole" and can probably use the existing ferrule/splice with all wires connected from one side. @formatc1702 should have this in mind when he plan to rename the connector category value.

formatc1702, 2020-07-06:
Like @kvid mentioned, both wire nuts and Wago 221 series connectors behave very much like a splice. Thus, they should be represented as such in WireViz.

> wires in Wago connectors are all on the same side of the connector. It would be nice if this important property could be carried over in the graphical representation.

For a functional representation of the wiring (which is what WireViz is for), this is irrelevant, and also hard to implement in the current code.

See example 05:
[image]
`X1`, `X2` and `X3` are all the same type of connector, but `X2` has wires coming out both sides of each pin due to the daisy chaining.

I will therefore close this issue and refer to #26 for now.

closed 2020-07-06 by formatc1702

formatc1702, 2020-07-06:
As a bonus, here's how the image in the first comment might be rendered in WireViz:
(PE rendered as green, since multicolored wires are still in the works)

[image]

I realize the text to generate the connections is not very compact, this is an issue that could be addressed in the future.

```yaml
cables:
  C1: &cable
    type: Mains cable
    colors: [BN, BU, GN] # GN to be replaced with GNYE for PE
    gauge: 1.5
  C2:
    <<: *cable
  C3:
    <<: *cable

connectors:
  S1: &wago
    category: ferrule
    manufacturer: Wago
    manufacturer_part_number: 221-413
    color: OG
  S2:
    <<: *wago
  S3:
    <<: *wago

connections:
  -
    - C1: 1
    - S1
  -
    - C1: 2
    - S2
  -
    - C1: 3
    - S3:
  -
    - C2: 1
    - S1
  -
    - C2: 2
    - S2
  -
    - C2: 3
    - S3
  -
    - S1
    - C3: 1
  -
    - S2
    - C3: 2
  -
    - S3
    - C3: 3
```