wireviz/WireViz #48: [feature] How to short multiple pins on one connector?

azivkovic3l, 2020-06-30:


formatc1702, 2020-06-30:
There is primitive functionality to attach two pins of the same connector. Please see the end of this modified example:

```yaml
connectors:
  X1:
    type: D-Sub
    subtype: female
    pinout: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]
  X2:
    type: Molex KK 254
    subtype: female
    pinout: [GND, RX, TX]

cables:
  W1:
    gauge: 0.25 mm2
    length: 0.2
    color_code: DIN
    wirecount: 3
    shield: true

connections:
  -
    - X1: [5,2,3]
    - W1: [1,2,3]
    - X2: [1,3,2]
  -
    - X1: 5
    - W1: s
  - # short pins 6,7,8 together
    - X1: [6,7]
    - X1: [7,8]
```

See the output below. It's not pretty, so I am open for suggestions on how to best visualize this. I guess there is also a difference between running a wire between two pins, and connectors that allow some other sort of shorting internally.

[image]

quotschmacher, 2020-07-01:
maybe it would be necessary to specify a starting and ending angle and a width, so that the shortage can be shorter?

azivkovic3l, 2020-07-01:
Thanks for quick reply,

Short term, it would be nice to include this option in examles so everibody
know about it.
Long term, consider shorter length of connection line.

Regards,
Aleksandar

formatc1702, 2020-07-01:
Yes, I do agree the current implementation is ugly and missing information... any ideas and/or pull requests on this topic are welcome!

quotschmacher, 2020-07-01:
my python is too weak to help i think.
what about a (overriding) color attribtue for the connection? overriding: override the cable color if a cable is used. and an array in preparation for multi-color wires.

`connections:
  - color: [RD]
    - Relais1: [6, 12, 18 ]
    - Relais1: [12, 18, 24]
`

formatc1702, 2020-07-10:
Here's a mockup of what it could look like:
[image]

I'm not sure how easy that would be to represent graphically... I'll think about it.
A possible input syntax could be:
```yaml
connectors:
  X1:
    pinout: [...]
    internal_shorts:
      - [1,2]
      - [3,5,7]
      - [4,6]
```

The primitive example from my comment above, with external loops, could be solved in a similar way:

```yaml
connectors:
  X1:
    pinout: [...]
    external_shorts:
      - [6,7] # exactly 2 pins must be specified
      - [7,8]
```

Only thing missing is a bit of info on what kind of wire to use for the external shorts.

azivkovic3l, 2020-08-19:
I need small help, I'm using your dev branch because of new features that are still not in master. I see that pins shorting feature is moved to version 0.3. Until than, dev branch complains with the following "Exception: X1 is not in cables" when I define that connector X1 must have shorted pins 3 and 10:
    - X1: [3,10]
    - X1: [10,3]

What is a new syntax for this shorting in dev branch?

formatc1702, 2020-10-20:
Sorry for the super late reply!
In the newly released v0.2, try the following to short pins 3 and 10... it will be displayed as a wire loop between them, and you have no control over the wire appearance/properties, but maybe this helps you.

```
X1:
  # connector parameters
  loops:
    - [3, 10]
```

dbozec, 2022-08-23:
Hi, I have a small issue with those loops.
On one of my project the loop wire is hidden behind the Wire description table next to the connector
[image]
It's not really readable :(
```yaml
  -
    - P1: [1]
    - W1: [1]
    - P1: [9]
```

formatc1702, 2022-08-25:
Yes, I think loops and internal shorts is something we should tackle soon, since a lot of users request this and complain about the current implementation... which is understandable :)

kvid, 2024-05-04:
@azivkovic3l, @quotschmacher, @dbozec, and others with similar use cases - I added today a suggested work-around for over-sized loops in https://github.com/wireviz/WireViz/issues/286#issuecomment-2094309143

dbozec, 2024-07-26:
Sorry for late reply @kvid :
your tweak works on simple cables, but can mess with wires in some complex ones :/

kvid, 2024-07-26:
@dbozec wrote:
> Sorry for late reply @kvid : your tweak works on simple cables, but can mess with wires in some complex ones :/

Please also try the same (complex) YAML input with the currently latest version 0.4.1 that hopefully should improve the output in such use cases.

dbozec, 2024-07-29:
Great, it's a lot better, thank you !