wireviz/WireViz #49: Add support for line breaks

MikeCrookes, 2020-07-01:
Would be great to be able to add line breaks into the fields as long descriptions can extend the output past normal screen width

formatc1702, 2020-07-03:
This sounds reasonable. Which fields in particular are you thinking of?
Surely the notes fields... any other? I wouldn't want the code to become too cluttered with too many `\n` to `<br>` conversions all over the place.

MikeCrookes, 2020-07-04:
I'm not that familiar with all the fields (and what they do) but looking at the data classes I think may be useful for;
- name
- category
- type
- subtype
- notes (as you suggested)

formatc1702, 2020-07-05:
I have started some work in the [`feature/linebreaks`](https://github.com/formatc1702/WireViz/tree/feature/linebreaks) branch.

I will add support for the following fields:
- `type`
- `subtype`
- `notes`

`name` is the unique designator for a component and should be short and sweet (e.g. `X1`), no need for line breaks.
`cagegory` is only used internally, currently to distinguish between connectors and ferrules, as well as between cables and wire bundles. Thus the user does not need to edit this field.

formatc1702, 2020-07-05:
Closed by #64.

PR #64 (2020-07-05): Add support for line breaks in certain fields
Closes #49. @Tyler-Ward perhaps you can can have a quick look and let me know if something's missing.

I am aware that ferrules are not rendered very nicely if their fields include line breaks (try my [test code](https://gist.github.com/formatc1702/531e68e3da2c180006808acdaf797cfe)) so I'll tackle that next. It's a cosmetic issue not related to the actual line break support.
closed 2020-07-05 by formatc1702