wireviz/WireViz #51: Manual BOM additions

MikeCrookes, 2020-07-01:
Nice feature would be to be able to manually (via the yml) add lines to the auto-generated BOM

Tyler-Ward, 2020-07-05:
Commenting so issue can be assigned to me.

Tyler-Ward, 2020-07-12:
A pull request for this has been opened in #80

A future expansion of this could be to add a list of extra parts for a connector or cable for example where a connector has a seperate backshell part or for things like spiral wrap, This would enable it to be templated as part of the connector or cable and for qty and designators to be calculated automaticaly.

formatc1702, 2020-07-13:
Closed. Feel free to open a new issue for adding custom BOM items within a connector! Sounds like a useful feature.
What about within a cable? Any situation where that might be useful?

closed 2020-07-13 by formatc1702

aakatz3, 2020-07-16:
Could be useful to have custom bom within a cable (or maybe a bundle?) for example, heat shrink, or sleeving