wireviz/WireViz #53: [feature] color mark at connector pins

quotschmacher, 2020-07-01:
add the ability to add a small color mark or something similar at connector pins.

formatc1702, 2020-07-02:
You mean each pin might have a color assigned to it? That would mean we would need a new optional list-type parameter with the same length as pinout (and optionally pinnumbers). This sounds like a good idea.

I was also thinking of the option to specify a color for the entire connector (just in case).

If anybody wants to take this on, please let me know. I don't think I will find time for this soon, myself.

Thanks for the request!

formatc1702, 2020-07-02:
Internal note: This (both the original suggestion and the ability to assign color to entire connector) will require rewriting the node's GraphViz code from a `record` (which is much more compact code-wise, but does not support colors inside cells AFAIK) to a proper HTML node (just like cables are already implemented)

formatc1702, 2020-07-06:
Update: Ferrules have been refactored in #65. Original issue still open.

formatc1702, 2020-07-06:
> Internal note: This (both the original suggestion and the ability to assign color to entire connector) will require rewriting the node's GraphViz code from a `record` (which is much more compact [...]

Update: Ferrules have been refactored in #65. Original issue still open.

PR #141 (2020-11-14): Add option to add colors to connector pins
Closes #53.
Based on `feature/gv-html-refactor` branch.

- Adds new `pincolors` list attribute to connectors.
  - If `pincolors` is used, needs to be the same length as `pins`, `pinlabels`
- If used, displays a small color mark and label next to each pin
  - Color mark is smaller than the `connector.color` mark on purpose
- Implements `__` as placeholder for pins with no color
  - Better suggestions welcome
  - Edit 2020-10-21: Any unknown color code is treated as "no color", so `__`, `--`, `XX` and anything else may be used.
- Makes sure white color  [... cut]
closed 2020-11-14 by formatc1702