wireviz/WireViz #57: ModuleNotFoundError: No module named 'dataclasses'

davidrea, 2020-07-03:
Ran a fresh install of WireViz on Pop!_OS (Ubuntu) 18.04 LTS following the instructions in README.md. When attempting to run WireViz (either on my own yml file or one of the demos from the repository) I get:

```
david@pop-os:~$ wireviz test.yml 
Traceback (most recent call last):
  File "/home/david/.local/bin/wireviz", line 11, in <module>
    load_entry_point('wireviz', 'console_scripts', 'wireviz')()
  File "/home/david/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/david/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2855, in load_entry_point
    return ep.load()
  File "/home/david/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2446, in load
    return self.resolve()
  File "/home/david/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2452, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/david/dev/WireViz/src/wireviz/wireviz.py", line 5, in <module>
    from dataclasses import dataclass, field
ModuleNotFoundError: No module named 'dataclasses'
```

I confirmed GraphViz is installed:

```bash
david@pop-os:~$ sudo apt install graphviz
Reading package lists... Done
Building dependency tree       
Reading state information... Done
graphviz is already the newest version (2.40.1-2).
```

Tried running the `pip3 install` command as root, same issue. Did I miss a step?

davidrea, 2020-07-03:
Just noticed this warning way down in the README.md file's hinterlands:

> Ubuntu 18.04 LTS users in particular may need to separately install Python 3.7 or above, as that comes with Python 3.6 as the included system Python install.

Installed Python 3.7 from source, then installed wireviz by calling pip3.7 explicitly, working fine now...

closed 2020-07-03 by davidrea

louiscklaw, 2020-07-21:
If someone in similar situation (ubuntu 18.04 + python 3.6), install `dataclasses` separately may help.

### command:
`sudo python3 -m pip install dataclasses`

Install it and the example "Demo01" on readme page can be generated on my side.

### Log:
```
$ sudo python3 -m pip install dataclasses
WARNING: The directory '/home/logic/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting dataclasses
  Downloading dataclasses-0.7-py3-none-any.whl (18 kB)
Installing collected packages: dataclasses
Successfully installed dataclasses-0.7

$ python3                                
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
KeyboardInterrupt
>>> 

$ uname -a
Linux logic-OptiPlex-9020 5.3.0-62-generic #56~18.04.1-Ubuntu SMP Wed Jun 24 16:17:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

$ wireviz ./wireviz_helloworld.yml

```