wireviz/WireViz #61: [feature] Add 1:1 "to-scale" printing output or image creation, for peg-board layout generation

tim292stro, 2020-07-04:
In higher volume production environments, a harness that will be produced several times is generally laid out on a peg board.  This allows wires to be cut in-place at the correct length, and laid from point-to-point before bundling and insertion into the connector.
[image]

One normally prints out a harness on a large format printer, then lays the diagram on an appropriately sized material.  This can be done on either a piece of wood or metal, normally pegs or nails can be inserted at fan-out points, allowing the wires to be separated.  This can also be done on a magnetic material and using hanger magnets (magnets with hooks attached).
[image]

For quality control purposes, these peg boards often have the mating connector for end terminations so that once completed the entire harness can be fault-checked.  Continuity checks

For a short-run/bespoke or doing several different harnesses per day in the same station, having a 1:1 scale model projected on a white metallic surface and using repositionable magnet hooks would allow the same effect without a large format printer.  Having a known scale in X/Y axis dimensions allows external math to be done.  For example having a 4k (3840x2160) projector throwing a 170.67"x96" 1:1 scale image on a white-painted steel table top we'd know each pixel was 0.044"x0.044".  That's a 14.25'x8' table.  This would require the image generation to consider the constraints of the output dimensions.  Note that on this example:
[image]
some legs of the harness are wrapped around to fit in the space allowed.

aakatz3, 2020-07-21:
This is a very interesting problem to solve. Are these usually created by hand, and how are the paths laid out? It might be possible, using the bundles and a new output stage for the harness, to generate paths with specific lengths, and "length tuning" to make sure the lengths take up minimal physical space.

tim292stro, 2020-07-21:
Normally these are done in CAD with a (rather expensive) 3rd party tool (like this from Mentor Graphics: "https://www.mentor.com/products/electrical-design-software/capital-harness-xc" - I believe it's ~$2K-USD/seat IIRC).  The tool takes into account the destination page size then wraps them if needed to splay them out without overlapping.  Think something like a big hydra-level game of "Snake".

Until recently, I had been using the side of a steel container and some hook magnets (like: https://www.amazon.com/BAVITE-Magnetic-Neodymium-Kitchen-Workplace/dp/B081WVM17B) to do layouts on a wall.  Just used a notepad and a tape measure, but the setup and QC checking time was a killer.

The critical detail is getting the lengths between the junctions and the ends of the bundles correct - then having the correct lengths for each connector wire (all of this is already captured in the text files from what I can read).

formatc1702, 2020-07-21:
While it does sound very interesting, I doubt we'll be seeing this feature in WireViz anytime soon... since GraphViz loves moving nodes around, and it's hard enough to get it to display nicely at all... it's optimized to display logical elements and connections among them, not a physical layout.
But, one can dream :)

tim292stro, 2020-07-21:
Yes, I realized that GraphViz is limiting or making it necessary to work around (more than just this item) - this is more of a wish-list I guess should one be considering porting another graphic back-end API.

aakatz3, 2020-07-24:
As a separate note, if you have Microsoft Visio, and GraphVisio, you may be able to take the wireviz output, and then possibly lay it out to-scale in that? Visio is expensive, and GraphVisio is somewhat buggy, but it is an option!