wireviz/WireViz #66: [internal] Refactor HTML generation for GraphViz output

formatc1702, 2020-07-05:
#65 introduced [a more elegant way](https://github.com/formatc1702/WireViz/pull/65/files#diff-f3f7c9cc77f143d3a653259c03d87d40R63-R82) to generate the HTML code to use inside the GraphViz nodes.

So far, it's only implemented for ferrules. I will implement similar changes for [non ferrule] connectors as well as for cables/wire bundles, as soon as #17 is merged (to avoid any further trouble while merging.

formatc1702, 2020-07-05:
Work for connectors has started in the [`feature/gv-html-refactor`](https://github.com/formatc1702/WireViz/tree/feature/gv-html-refactor) branch.

aakatz3, 2020-07-24:
Should likely be implemented with part of #123 (noted in discussion)

formatc1702, 2020-07-24:
See my comment there.

> In order not to further complicate things, I'd like to move the HTML output as-is to this new module... the HTML itself is a story for another day (#74).

PR #136 (2020-08-13): Feature/gv html refactor
Fixes #66.

Adds a `color` property to cables, to match the behavior of connectors, for consistency.

This might be useful, for example, for highlighting cables that remain live after an emergency stop or mains disconnection, that need to be orange-coloured according to certain standards ([Source](https://www.elektropraktiker.de/nc/fachartikel/leiterkennzeichung-in-der-farbe-orange/) in German).
closed 2020-08-13 by formatc1702