wireviz/WireViz #69: [bug] fix issue when hiding both name and wireocout of a cable/bundle

formatc1702, 2020-07-09:


kvid, 2020-07-20:
Please include a YAML file here that demonstrates the bug. That would make it easier to verify the fix, and it should be included in the corner cases described in issue #63.

formatc1702, 2020-07-20:
```yaml
connectors:
  X1:
    type: D-Sub
    subtype: female
    pincount: 9
  X2:
    type: Molex KK 254
    subtype: female
    pincount: 3

cables:
  W1:
    wirecount: 3
    show_name: false
    show_wirecount: false
    # shield: true

connections:
  -
    - X1: [5,2,3]
    - W1: [1,2,3]
    - X2: [1,3,2]
```

kvid, 2020-07-20:
I can verify the bug using the [current dev](https://github.com/formatc1702/WireViz/tree/a418005c2e101b35da00ca6e6b277450323c3577):
```
(wv-env) kvid@PC:~/src/formatc1702/WireViz/src/wireviz$ python wireviz.py _test/issue69.yml
Error: syntax error in line 1
... <tr></tr> ...
in label of node W1
Warning: node W1, port w1 unrecognized
Warning: node W1, port w2 unrecognized
Warning: node W1, port w3 unrecognized
Warning: node W1, port w1 unrecognized
Warning: node W1, port w2 unrecognized
Warning: node W1, port w3 unrecognized
Traceback (most recent call last):
  File "wireviz.py", line 252, in <module>
    main()
  File "wireviz.py", line 248, in main
    parse(yaml_input, file_out=file_out)
  File "wireviz.py", line 178, in parse
    harness.output(filename=file_out, fmt=('png', 'svg'), view=False)
  File "../wireviz/Harness.py", line 294, in output
    graph.render(filename=filename, view=view, cleanup=cleanup)
  File "/home/kv/src/formatc1702/WireViz/wv-env/lib/python3.7/site-packages/graphviz/files.py", line 209, in render
    quiet=quiet)
  File "/home/kv/src/formatc1702/WireViz/wv-env/lib/python3.7/site-packages/graphviz/backend.py", line 221, in render
    run(cmd, capture_output=True, cwd=cwd, check=True, quiet=quiet)
  File "/home/kv/src/formatc1702/WireViz/wv-env/lib/python3.7/site-packages/graphviz/backend.py", line 184, in run
    output=out, stderr=err)
graphviz.backend.CalledProcessError: Command '['dot', '-Tpng', '-O', 'issue69']' returned non-zero exit status 1. [stderr: b'Error: syntax error in line 1 \n... <tr></tr> ...\nin label of node W1\nWarning: node W1, port w1 unrecognized\nWarning: node W1, port w2 unrecognized\nWarning: node W1, port w3 unrecognized\nWarning: node W1, port w1 unrecognized\nWarning: node W1, port w2 unrecognized\nWarning: node W1, port w3 unrecognized\n']
```
[image]

PR #104 (2020-07-20): Fix node rendering for cables with hidden name and/or no visible attributes
Closes #69.

Probably just an interim solution until work on #66 continues, and a more elegant fix arises.
commit b028e7a2 (2020-07-20): Fix node rendering for cables with hidden name and/or no visible attributes (#104)

Closes #69.
closed 2020-07-20 by formatc1702