wireviz/WireViz #72: [feature] Throw error if a connector has duplicate entries in the pinnumbers list

formatc1702, 2020-07-09:
#10 allows the user to specify custom pin numbering schemes instead of auto-assigning pins 1 through N.

A check should be performed whether any pin number is assigned twice, and throw an error if so.

This is different from #71, since pin _names_ need only be checked for duplicates if actually attempting to connect a pin with an ambiguous name, whereas pin numbers should be unique by definition, so this check should be performed during connector creation.

closed 2020-07-09 by formatc1702

kvid, 2020-07-09:
Do you also check for any collision with `s` - the shield identifier? See my earlier comment in https://github.com/formatc1702/WireViz/pull/10#issuecomment-650585648

formatc1702, 2020-07-09:
Thanks for reminding me, I never replied to that comment.
It is not an issue, since we are talking about _connector_ pins/ports, whereas the `s` identifier is a _cable's_ port. The GraphViz format for linking to a specific port inside a node is `node_name:port_name` so there won't be any confusion.

kvid, 2020-07-09:
OK, I thought these consepts applied to both cables and connectors. There exist connectors with shield that is not included in the pin numbering. And why not support custom wire identifiers also for completeness or to generalize the 'syntax? These are just my thoughts. I have no real need for any of it at the moment.

formatc1702, 2020-07-09:
> There exist connectors with shield that is not included in the pin numbering.

Yes, but right now, only cables have a dedicated `shield: true` property, which auto-generates a pseudo-wire labeled `s`. In the future, we will have to think about restructuring shields, to support more complex schemes such as multiple shields, nested shields, etc.

> And why not support custom wire identifiers also for completeness or to generalize the 'syntax?

Indeed, I also want to add a `wirenames` property to cables/bundles, and also make it possible to link to a specific wire color, as long as it's unambiguous.

kvid, 2020-07-10:
I'm sorry for all my comments about issues that don't apply. I don't have full knowledge about the source, and make wrong assumptions now and then. I'm impressed that you take the time to answer most comments this quickly.

formatc1702, 2020-07-10:
I'm really thankful for the open discussions, improvement suggestions and code reviews!
If I didn't want that, I'd take the project off GitHub ;)

I want to follow through with as many features and bugfixes as possible now that momentum -both on my side and in the contributors- is high, and I can afford to put my time into the project :)

commit 6a2924fa (2020-07-09): Check for duplicate pin numbers and raise an exception

Closes #72.