wireviz/WireViz #85: [info] Docstring and typehint guidance

slightlynybbled, 2020-07-14:
Would like guidance from author with regard to docstring and type hint format.

I have been messing with the base `Harness` class to create the GUI with which you are already familiar.  I am finding several ambiguities that I would like to have cleared up, but I don't know the docstring format which you would like to start with.

Generally, I use the pycharm-suggested docstrings:

    def connect(self, from_name: str, from_pin: (int, str), 
                       via_name: str, via_pin (int, str),
                       to_name: str, to_pin: (int, str)):
    """
    Add a connection from one connector/pin to another connector/pin.
    
    :param from_name: the cable name for the starting pin
    :param from_pin: and integer or the string "s" for shields
    :param .....
    """
        pass

formatc1702, 2020-07-14:
I am aware that documentation is scarce ATM, and it's something that should be worked on.

I am not yet familiar with docstrings, and don't know what options there are. I will need some time to familiarize myself with different alternatives. If the current contributors have any suggestions, please let me know.

If there are any specific questions regarding ambiguities, feel free to sumbit an issue or contact me directly; I'll try my best to reply quickly but can't promise anything.

slightlynybbled, 2020-07-15:
[Some docstring reading](https://realpython.com/documenting-python-code/)

I tend to use the "reStructuredText" method.  Pycharm generally recommends this and I'm a defaults guy.  It also turns out to be more compact than some of the others.

The google and numpy conventions are just as valid and well-supported by [Sphinx](https://www.sphinx-doc.org/en/master/).

Whatever you choose, so long as it is compatible with document generation tools such as Sphinx, you can't really go wrong.  You just want to be consistent.

formatc1702, 2020-07-15:
Thanks for the link.
I have to say, the Google Docstrings are the easiest to visually parse at a glance; IMHO reStructured Text is too dense.
So I'd go with Google. (Sorry, the only existing docstring for `wireviz.parse()` will have to be changed 😉 ) 

That said, I won't have time to sit down and write nice docs in the upcoming days. I appreciate and welcome any contributions, though. And like I said, if there are any points that need clarification, I'm happy to help.

slightlynybbled, 2020-07-15:
Ok, thanks.  I just don't want to waste time doing it if it is going to be the wrong format.

formatc1702, 2020-07-15:
Thanks for pushing me to at least do *some* documentation ;-)

closed 2020-07-15 by slightlynybbled