wireviz/WireViz #86: \n works in connector notes, not

davidrea, 2020-07-14:
Adding a `\n` escaped character works to insert a line break in:

* Connector `notes:` entities
* Connector `type:` and `subtype:` entities

But the `\n` escaped character is printed in cable `notes:` entities:

[image]

Source for above:

```yml
connectors:

  P1:
    type: Socket housing
    subtype: Molex SL\n50579406
    pinout: [VDD, SWCLK, GND, SWDIO, NRST, SWO]
    notes: "Terminal: 16020086 or equiv."

  P2:
    type: Pin header
    subtype: Adam Tech\nBHR-20-VUA\nor Equiv.
    pinout: [VCC, VCC (optional), TRST, GND, TDI, GND, DIO/TMS, GND, DCK/TCK, GND, RTCK, GND, SWO/TDO, GND, NRST, GND, NC, GND, NC, GND]
    notes: Solder wires to pins,\nheat shrink individually

cables:
  W1:
    gauge: 24 AWG
    length: 0.2
    wirecount: 6
    category: bundle
    colors: [RD, GY, BK, GY, GY, GY]
    notes: Stranded,\nHeat shrink bundle 2 plcs
    # ^^^ this line contains printed \n ^^^

connections:
  -
    - P1: [1, 2, 3, 4, 5, 6]
    - W1: [1, 2, 3, 4, 5, 6]
    - P2: [1, 9, 4, 7, 15, 13]
```

davidrea, 2020-07-14:
Just tacking onto this, \n escaped characters also appear in HTML BOM:

[image]

formatc1702, 2020-07-14:
Thanks for the heads up. I can't look at the issue in detail now, but I can offer solution that (IMHO) is more elegant than using `\n` for new lines, since [YAML supports multiline strings natively](https://yaml-multiline.info/).
The code below fixes both the rendering and BOM issues:

```yaml
connectors:
  P1:
    type: Socket housing
    subtype: |
      Molex SL
      50579406
    pinout: [VDD, SWCLK, GND, SWDIO, NRST, SWO]
    notes: "Terminal: 16020086 or equiv."

  P2:
    type: Pin header
    subtype: |
      Adam Tech
      BHR-20-VUA
      or Equiv.
    pinout: [VCC, VCC (optional), TRST, GND, TDI, GND, DIO/TMS, GND, DCK/TCK, GND, RTCK, GND, SWO/TDO, GND, NRST, GND, NC, GND, NC, GND]
    notes: |
      Solder wires to pins,
      heat shrink individually

cables:
  W1:
    gauge: 24 AWG
    length: 0.2
    wirecount: 6
    category: bundle
    colors: [RD, GY, BK, GY, GY, GY]
    notes: |
      Stranded,
      Heat shrink bundle 2 plcs

connections:
  -
    - P1: [1, 2, 3, 4, 5, 6]
    - W1: [1, 2, 3, 4, 5, 6]
    - P2: [1, 9, 4, 7, 15, 13]
```

I will still keep this open, but I'm unsure whether converting `\n` to newline is something that should be implemented if there is a more standard way in YAML.

formatc1702, 2020-07-14:
Looking at the `Flow Scalars` section on [yaml-multiline.info](https://yaml-multiline.info/), it seems that _not_ converting `\n` into a newline for plain strings is the expected behavior. I can't say why it does work in connectors and not in cables; I'd attribute it to a quirk in the `pyyaml` module.

kvid, 2020-07-15:
WireViz does not interprete the `\n` escape sequence, and sends it untouched to the output files, including the .gv file that is read by Graphviz. However, Graphviz does interprete [several escape sequences](https://graphviz.org/_pages/doc/info/attrs.html#k:escString) and that is why the `\n` split the line in connector attributes that use `shape=record`. The wire nodes are formatted with [HTML](https://graphviz.org/doc/info/shapes.html#html) and the `\n` escape sequence is apparently not interpreted then (`<BR/>` is used to break the line in HTML).

kvid, 2020-07-15:
When testing the better approach with multi-line YAML, I became aware of a minor inconsistency about horizontal alignment that I raised as the new issue https://github.com/formatc1702/WireViz/issues/90

davidrea, 2020-07-15:
Thanks for the feedback on yaml, I wasn't aware of that! I'll switch to using the standard multiline input on this diagram.

closed 2020-07-20 by formatc1702