wireviz/WireViz #90: [bug] Inconsistent horizontal alignment

kvid, 2020-07-15:
Single-line connector attributes are centered horizontally when smaller than the available space, but the `graphviz_line_breaks()` function make multi-line attributes left aligned:
https://github.com/formatc1702/WireViz/blob/34cd9c26126231eeb25f93af245b0a24e8d45ed8/src/wireviz/wv_helper.py#L120

kvid, 2020-07-15:
Some users might argue that they prefer left-aligned multi-line attributes. IMHO, that would be a new feature request.

formatc1702, 2020-07-15:
I intentionally used `\l` because I thought it would look better, but you are right, centered looks more consistent. I'll think about adding alignment options later, but it's not a priority.

closed 2020-07-15 by formatc1702