wireviz/WireViz #97: [bug] Improper XML and DOCTYPE Declarations

kvid, 2020-07-17:
The HTML output violates these rules:
- [If the XML declaration is included, it must be situated at the first position of the first line in the XML document.](https://xmlwriter.net/xml_guide/xml_declaration.shtml)
- [The document type declaration must be placed between the XML declaration and the first element (root element) in the document.](https://xmlwriter.net/xml_guide/doctype_declaration.shtml)

The root cause is that `Harness.output()` start HTML output with [`file.write('<html><body style="font-family:Arial">')`](https://github.com/formatc1702/WireViz/blob/861380ddd64082a6b478f3d8acf8680ae3b67356/src/wireviz/Harness.py#L307) (i.e. no leading XML or DOCTYPE declarations) and then later embed the SVG diagram into the HTML output without hiding (remove or comment) the leading XML and DOCTYPE declarations from the SVG file.

If we are including XML and/or DOCTYPE declarations, then we should comply to their rules. Related issues I have not investigated:
- Does all the HTML generated by `Harness.output()` comply to identical XML or DOCTYPE declarations as the SVG file?
- Can a proper XML declaration with `encoding="UTF-8"` be of any help in issue https://github.com/formatc1702/WireViz/issues/52 ?

aakatz3, 2020-07-17:
The svg doctype is generated by graphviz itself. The best way to deal with this would be a complete refactor of HTML generation, using a library. I'm not amazing at python, but I found a couple that can be tried.

kvid, 2020-07-17:
> The svg doctype is generated by graphviz itself.

I know that, but it should be easy to filter out the leading declarations before embedding the rest of the SVG file.

> The best way to deal with this would be a complete refactor of HTML generation, using a library. I'm not amazing at python, but I found a couple that can be tried.

I cannot see how a library will solve this issue. It might be a good idea for other reasons, but that belong in a separate issue. If we want to add proper leading declarations, then it is easy to include in the first `file.write()`. IMHO, "a complete refactor of HTML generation" is overkill in this case.

The question is: Do we want any of these declarations at all?

aakatz3, 2020-07-17:
My thought was that if we use the xml.dom.minidom library, we can easily strip out or modify tags as necessary from the Graphviz SVG. Similarly, if we use that library, or a similar one for HTML, we can then just pass an object instead of repeatedly reading and writing files. 

The doctype and meta tags are probably pretty important for W3C compliance [1](https://www.w3.org/QA/Tips/Doctype) [2](https://www.w3.org/QA/2002/04/valid-dtd-list.html), if we care about that.

kvid, 2020-07-18:
The [W3C Markup Validation Service](https://validator.w3.org/) complained with these errors:
- When using XML declaration: Saw `<?`. Probable cause: Attempt to use an XML processing instruction in HTML. (XML processing instructions are not supported in HTML.)
- When not using DOCTYPE declaration: Start tag seen without seeing a doctype first. Expected `<!DOCTYPE html>`.
- When SVG DOCTYPE declaration was not removed: Stray doctype.
- Missing and obsolete elements in the HTML, but that belongs in a separate issue.

formatc1702, 2020-07-18:
Thanks for the PR, this fixes the primary issue.

The other minor issues should be addressed as part of #32 and the [feature/technical-drw](https://github.com/formatc1702/WireViz/tree/feature/technical-drw) branch.

closed 2020-07-18 by formatc1702