wireviz/WireViz #101: [meta] Collaborator discussion

formatc1702, 2020-07-19:


formatc1702, 2020-07-19:
I've created [a new milestone](https://github.com/formatc1702/WireViz/milestone/1) to show which features I'd like to have included in the next release, i.e. pushing to `master` as well as uploading the new package to [PyPI](https://pypi.org/project/wireviz/).

If anybody wants to suggest more issues to include, please let me know. However, my aim is to release v0.2 relatively soon, which is why I've deliberately left out a lot of existing requests.

formatc1702, 2020-07-21:
Uptate on the [v0.2 milestone](https://github.com/formatc1702/WireViz/milestone/1):
I've assigned people for most open issues/PRs, hoping to be able to assign #114 soon too. If anybody wants to take on #60, please comment there and I will asign you.

Additionally, I've made a [milestone for v0.3](https://github.com/formatc1702/WireViz/milestone/2), which currently only serves to denote issues/PR drafts that I know I want to pursue as soon as v0.2 is out, but not before.

kvid, 2020-07-24:
Have you considered #116 and #121 for the [v0.2 milestone](https://github.com/formatc1702/WireViz/milestone/1)?

formatc1702, 2020-07-24:
Added both! Thanks.

kvid, 2020-08-01:
I just made a suggestion in the closed issue https://github.com/formatc1702/WireViz/issues/122#issuecomment-667545193. Do you think it should be reopened?

formatc1702, 2020-08-01:
Feel free to do so!

kvid, 2020-08-01:
> Feel free to do so!

I don't know how. Maybe I'm not authorized to do it.

formatc1702, 2020-08-01:
Not sure.. did it myself now :)

kvid, 2020-08-01:
> Not sure.. did it myself now :)

Thank's

> 
> Perhaps only repo owners (and maybe the original commenter) have the "reopen issue" button...

You are right. I certainly did not see that button here, but I can see it in one of the closed issues I opened myself.

formatc1702, 2020-08-04:
Just a quick FYI especially for @kvid, I'll be taking a couple of days off coding to breathe a bit, and I promise I shall be back to read through all the comments, commits and other contributions as usual.
It will help me get back my focus and ensure v0.2 gets released with no major issues :)

formatc1702, 2020-08-17:
My goal is to get [v0.2](https://github.com/formatc1702/WireViz/milestone/1) released by the end of the week.
- #27 / #153 is almost ready for merging
- #107 / #111 needs to be updated to reflect the above PR and is then ready, too.
- #60 / #98 / #123 have been postponed to v0.3 due to inactivity.
  I'm assuming this is due to the holiday period and I hope the authors make a comeback at some point :)
  Otherwise, I'm sure we'll find a way to work on this while keeping the authorship intact.

"Perfect is the enemy of the good", so let's get this release out soon and hopefully receive feedback on the new features. I suggest that after the release, we can decide on prioritizing the upcoming issues to get another cool set of features for v0.3!

Thanks everybody who's contributed so far!

kvid, 2020-10-04:
When the PRs are ready and unless rejected by @formatc1702 , I suggest he merge them into `dev` in this order: #115, #164, #163, #153, and #111, but please give the PR authors a couple of days for rebasing on top of the new `dev` and resolving conflicts before each merge.

formatc1702, 2020-10-05:
Thanks for the suggestion. I haven't looked at the contributions in detail; just one question:

Why not prioritize #153 and #111, which were planned for v0.2 already, get that one released, and then continue with the other PRs you mention? I'm worried that the v0.2 release will be further delayed if we try to include every new addition, and it might be better to make a clean cut between what goes into v0.2, and what will go into v0.3 and beyond.

If there's a technical reason that would prevent this, please let me know. Thanks!

kvid, 2020-10-05:
> Thanks for the suggestion. I haven't looked at the contributions in detail; just one question:
>
> Why not prioritize #153 and #111, which were planned for v0.2 already, get that one released, and then [...]

I see your point.

> 
> If there's a technical reason that would prevent this, please let me know. Thanks!

- There are no critical reasons that I know about. My line of thought was this: As many more PRs are ready or close to ready now compared to before your vacation, I thought perhaps you considered moving more of them into v0.2. #111 already describes `ignore_in_bom` and @Tyler-Ward suggested you cold move that part from his PR #115 into v0.2 - which is not a bad idea, but then during the last few days, the whole #115 has moved very close to ready, and I thought you would be tempted to move the whole #115 into v0.2.
- Given my assumptions, I suggested #115 first, since that one include a major refactoring of BOM generation that will benefit other PRs.
- However, as my assumptions that you maybe wanted to move new PRs into v0.2 was wrong, then #153 and #111 should be merged in first, but I hope you consider #115 soon rather than later. 😃 
- #163 is kind of coupled with a #111 suggestion, but it should be possible to accept my https://github.com/formatc1702/WireViz/pull/111#discussion_r487467403 even before #163 is merged in.
- #164 is quite small, but potentially very useful for many users.

formatc1702, 2020-10-05:
Thanks for the detailed answer.
- I will stick to finishing v0.2 with #153 and #111 first, to avoid "feature creep".
- I will then follow your other suggestions, and I see no reason not to release v0.3 soon afterwards... Since you've had a good eye on the project recently, I'm sure you can help pick out the PRs that are close to completion, and any burning issues that should be resolved and included in v0.3 as well.

There's tons of new stuff in v0.2 already, compared to the current master branch, so it's nice to keep some good stuff for the following release ;)

stevegt, 2020-10-05:
@formatc1702  Let us know how we can help.  I'm extremely impressed by how motivated and active the contributors to this project have been in its short life, and I'd like to do what I can to ensure its continuity -- I'm already pretty sure the WireViz yaml schema could become a defacto industry standard, though that does raise the thorny issue of governance -- maybe some sort of RFC or PEP-like approach might work.  I'm also thinking of suggesting you get on the waiting list for the github Sponsors program if you haven't already.

formatc1702, 2020-10-16:
> @formatc1702 Let us know how we can help. I'm extremely impressed by how motivated and active the contributors to this project have been in its short life, and I'd like to do what I can to ensure [...]

Thanks @stevegt for the kind words! It has been amazing to see the project grow, and indeed, collaboration with all the contributors has been great!

I'll lookinto the GitHub Sponsors thing; however, I'm unsure whether monetizing the project in this way might negatively impact users' expectations, and contributors' attitudes, since a lot of WireViz is built on their work, not only mine.

formatc1702, 2020-10-16:
> Let us know how we can help. 

Since my time for working on the project has been a bit limited lately, the following things have helped immensely:
- Contributors submitting high quality PRs, possibly rebasing onto `dev` if necessary, so all I need to do is nitpick a bit and click Merge ;)
- Contributors collaborating without my intervention and checking each others work (e.g. in #115), leading to the aforementioned quality
- @kvid keeping an overview of the various issues, how they interact, and suggesting an order in which to include them; as well as meticulously checking for uncaught mistakes and ensuring completeness of documentation, etc.

This is all highly appreciated, and I hope we can keep the momentum of the project going :)

formatc1702, 2020-10-17:
Posting it here as well for completeness sake:
v0.2 has been released, thank you everybody for your valuable contributions!

>100 forks and >2000 stars on the project really tell me we're working on something worthwhile here!

formatc1702, 2020-11-14:
Short summary of the current situation:

- #192 seems to be working, will re-review it soon and merge
- I am currently working on #186 and hope to have a version that is ready for review soon-ish
- #181 has been updated to reflect the recently merged PRs
- **I could use some help** with the following:
  - Getting #171 ready for merge (high priority, has been waiting for a while)
  - Getting #168 ready for merge
  - #189 might spawn a new PR for embedding images in the SVG/HTML output
    - I could tackle that after finishing the above PRs, but I will also accept contributions if anybody feels up for the task

As you can see, I'd like to reduce the number of open PRs first. Afterwards, I'll try and prioritize some issues that deserve attention :-)

kvid, 2020-11-15:
> Short summary of the current situation:
>
> * #192 seems to be working, will re-review it soon and merge
> * I am currently working on #186 and hope to have a version that is ready for review soon-ish
> * [...]

I have applied the commits from #171 + bugfix onto the current #192 commits and are ready to rebase it on top of `dev` and create a PR after #192 is merged in.

>   * Getting #168 ready for merge

I have done some work on this as well, but need more time because I'm not able to contribute every day.

formatc1702, 2020-11-15:
Thanks. #192 is now merged; looking forward to merging your upcoming PR and closing all the related partial PRs.

amotl, 2021-01-10:
Hi @formatc1702, @kvid and @stevegt,

within #211, we just reported about our efforts in bringing WireViz to the Web. It has actually just been a (re)packaging operation of code already conceived within the community. However, we hope you will appreciate having this available as a standalone package and we are looking forward to any contributions and comments.

With kind regards,
Andreas.

amotl, 2021-01-11:
Hi @formatc1702 and @kvid,

we just recognized that many improvements have been made on behalf of [1] and [2] (#197) already. Are you still planning to integrate those into mainline and release a version 0.3?

Just let me know when you are ready and I will update `wireviz-web` accordingly.

With kind regards,
Andreas.

[1] https://github.com/formatc1702/WireViz/tree/dev
[2] https://github.com/kvid/WireViz/tree/simplify-BOM-code

formatc1702, 2021-08-22:
Sorry for the lack of activity.

Here's the current state of things:
- As you can see, I have not had the time / mental capacity to take care of the project as much as I had hoped; however:
- I have been using WireViz extensively at work, using a "hacked together" branch that implements #74, #186 & #234 and I am very pleased with the results. It's nice to be able to use WireViz in a production environment.
- I hope to be able to merge these PRs into `dev` soon, to reduce the number of open PRs.
- Some of the other simpler PRs should follow (#197, #214, #219, #229)

I cannot promise a time frame for this, but I am hopeful that I can start the project moving again in the foreseeable future ;-)

Thanks everyone for your contributions.

formatc1702, 2021-08-22:
@kvid if you could let me know what order the mentioned PRs should be merged, and whether you need to do anything on them, that would help me to integrate them in the most painless way possible :) Thanks.

kvid, 2021-08-23:
As #197 is already merged, #214 and #219 can follow in that order. I have rebased them both on top of the current `dev`. Please review them to increase the quality. It should be possible to merge #229 both before or after them, but I have not checked for conflicts. I experience a lot of conflicts with import statements while rebasing. I think we should consider splitting imports into several lines (one identifier per line) and sort them alphabetically per import statement to reduce conflicts. Maybe there is a PEP describing this?

kvid, 2021-08-24:
I think this project follows [PEP8 import guidelines](https://www.python.org/dev/peps/pep-0008/#imports) more or less, but I think we should also consider sorting imports according to some common rules to reduce merge conflicts. Maybe a tool like [isort](https://pypi.org/project/isort/) can be used to automate the task? Some projects also combine isort with [black](https://pypi.org/project/black/) to force a fully unified code formatting on all Python files. Please tell about your own experiences with formatting rules and tools like this.

formatc1702, 2021-10-05:
It will soon be a year since v0.2 of WireViz was released. There are some rather big PRs pending, but I think that the `dev` branch has been in a very good shape recently (the only known major bug is #220, and a solution is on the way) so I have decided I will release v0.3 soon as it is in the current state because it includes some important features that I don't want to hold back any longer.

formatc1702, 2021-10-05:
>  I experience a lot of conflicts with import statements while rebasing. I think we should consider splitting imports into several lines (one identifier per line) and sort them alphabetically per import statement to reduce conflicts. Maybe there is a PEP describing this?

> I think we should also consider sorting imports according to some common rules to reduce merge conflicts

Recently, I have tried to stick to the following scheme and am trying to make this consistent across files.
Packages sorted alphabetically per block, and individual modules sorted alphabetically within a package's line.
```yaml
# built-in packages
import os
from pathlib import Path
import sys
# ...
        # blank line
# external dependencies
from graphviz import Graph
import yaml
# ...
        # blank line
# internal modules
from wireviz.DataClasses import Connector, Cable, Metadata, Options, Tweak
from wireviz.wv_colors import get_color_hex, translate_color
# ...
```

There will surely be some conflicts in the near future, but IMHO it is a clean and structured guideline that should reduce conflicts in the future.

amotl, 2021-10-05:
Dear Daniel,

we can warmly recommend using the excellent `isort` and `black` programs from the Python ecosystem to bring the codebase into a standards-based shape on those matters.

With kind regards,
Andreas.

formatc1702, 2021-10-08:
Thanks @kvid and @amotl for the `isort` and `black` suggestions! I am trying them out and I have to say, they look really promising!
Do you have experience with PRs branching off before reformatting with `black`, and attempting to merge after reformatting? Is it reasonable to expect that PRs will go smoothly if both the base branch and the PR branch are reformatted independently?

amotl, 2021-10-11:
Hi again,

I just wanted to leave another note on the `isort`/`black` thing. With https://github.com/earthobservations/gribmagic/commit/0d795ea, I just recently added it to another project and found the `--profile=black` option of `isort` sensible.

```
isort --profile=black foobar
black foobar
```

This option is important to make `isort`'s formatting of imports compatible with Black's code-formatting style.

> Do you have experience with PRs branching off before reformatting with black, and attempting to merge after reformatting? Is it reasonable to expect that merging will go smoothly if both the base branch and the PR branch are reformatted independently?

If both branches have been formatted with the same options, I believe it is reasonable to expect that merges will apply well without further ado.

With kind regards,
Andreas.

[1] https://pycqa.github.io/isort/docs/configuration/black_compatibility.html

martonmiklos, 2022-03-17:
Hi folks,

I am trying to get the #189  working. 
What is the status of the latest branch?

I got the following error:
```
mm@mm-precision:~/Dokumentumok/BlackEye/loopback/drawings/WireViz$ wireviz ../h/h.yaml 

WireViz 0.4-dev
Input file:   ../h/h.yaml
Output file:  ../h/h.[html|png|svg|tsv]
Traceback (most recent call last):
  File "/home/mm/.local/bin/wireviz", line 11, in <module>
    load_entry_point('wireviz', 'console_scripts', 'wireviz')()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/mm/Dokumentumok/BlackEye/loopback/drawings/WireViz/src/wireviz/wv_cli.py", line 137, in wireviz
    wv.parse(
  File "/home/mm/Dokumentumok/BlackEye/loopback/drawings/WireViz/src/wireviz/wireviz.py", line 88, in parse
    yaml_data, yaml_file = _get_yaml_data_and_path(inp)
  File "/home/mm/Dokumentumok/BlackEye/loopback/drawings/WireViz/src/wireviz/wireviz.py", line 397, in _get_yaml_data_and_path
    raise e
  File "/home/mm/Dokumentumok/BlackEye/loopback/drawings/WireViz/src/wireviz/wireviz.py", line 389, in _get_yaml_data_and_path
    yaml_path = Path(inp).expanduser().resolve(strict=True)
  File "/usr/lib/python3.8/pathlib.py", line 1181, in resolve
    s = self._flavour.resolve(self, strict=strict)
  File "/usr/lib/python3.8/pathlib.py", line 363, in resolve
    return _resolve(base, str(path)) or sep
  File "/usr/lib/python3.8/pathlib.py", line 347, in _resolve
    target = accessor.readlink(newpath)
  File "/usr/lib/python3.8/pathlib.py", line 452, in readlink
    return os.readlink(path)
OSError: [Errno 36] File name too long: "/home/mm/Dokumentumok/BlackEye/loopback/drawings/WireViz/ \nconnectors:\n  'aand here comes the YAML content
```

It looks that some paramteres swapped. Is it worth troubleshooting it or is there a better branch to start with the SVG embed?

jakew009, 2022-04-10:
Same issue as @martonmiklos

kvid, 2022-04-20:
I've not tested the #189 code myself after it was rebased on top of `latest`. The https://github.com/formatc1702/WireViz/pull/189#issuecomment-944955954 explains that there is an alternative branch `[backup/bugfix/image-src_before-latest-rebase](https://github.com/formatc1702/WireViz/tree/backup/bugfix/image-src_before-latest-rebase)`

amotl, 2022-08-05:
Dear Daniel,

it is sweet to see you working on #251 ff. again. Just let me know when you converged this into a new release, and I will update [WireViz-Web](https://github.com/daq-tools/wireviz-web) correspondingly.

With kind regards,
Andreas.

kvid, 2023-11-04:
My contributions to help with filalizing #251 is work in slow progress due to an increased effort I have to use on other important stuff in my life this autumn. Hence, I propose to consider creating a release 0.4 from the current `dev` branch - without #251 - to let user try out the new functionality now. It has already been too long since our last release.

I've already made an effort on this code by merging in a few extra PRs after the huge effort by @formatc1702 this summer, and fixing some minor stuff, so generating examples should work now.

I encourage @amotl and also other motivated users/contributors to try out the current `dev` branch and look out for minor things that are easy to fix before a release, like e.g. any missing or wrong documentation of new features. I also consider merging #298 into `dev` before this release, as it is a low hanging fruit and should be easy to integrate with #251 later.

kvid, 2024-01-21:
IMHO, we need to fix #328 before a new release. Please comment my suggested fix in that issue.

formatc1702, 2024-04-16:
I have submitted #332 as a response to #328.

I would like to go forward with merging #332 and #298 into `dev`, publishing v0.4 and afterwards immediately rebasing `big-refactor` onto `dev` and fast-fowarding, so finally we have a `dev` branch with more manageable PRs again.

formatc1702, 2024-04-16:
I have merged #332 and #298, updated the version number and added a [v0.4 tag](https://github.com/wireviz/WireViz/tree/v0.4).

However, I consider this an interim version that will not be merged into master or published on PyPI, and the [changelog](https://github.com/wireviz/WireViz/blob/v0.4/docs/CHANGELOG.md#04-unreleased) reflects this.

The next immediate step is rebasing #251 onto the latest `dev` branch, now labeled `v0.5-dev` to allow differentiating between changes from before (v0.4) and after (v0.5) the refactoring. I would appreciate any help in doing so. Then `dev` can be moved forward to contain the refactored code to finally get rid of `big-refactor`.

kvid, 2024-04-25:
> I have merged #332 and #298, updated the version number and added a [v0.4 tag](https://github.com/wireviz/WireViz/tree/v0.4).

Thank you!

> 
> However, I consider this an interim version that will not be merged into master or published on PyPI, and the [changelog](https://github.com/wireviz/WireViz/blob/v0.4/docs/CHANGELOG.md#04-unreleased) reflects this.

I don't understand why v0.4 should not be merged into master and released the normal way. My main motivation for suggesting v0.4 before merging #251 was to make it available to users and getting more usage feed-back.

It shouldn't be much work taking this remaining step to a full release, and I see no drawbacks about it, unless you know about any critical bugs or documentation flaws in v0.4. 

> 
> The next immediate step is rebasing #251 onto the latest `dev` branch, now labeled `v0.5-dev` to allow differentiating between changes from before (v0.4) and after (v0.5) the refactoring. I would appreciate any help in doing so. Then `dev` can be moved forward to contain the refactored code to finally get rid of `big-refactor` even if not all kinks are fully ironed out yet.

I agree to this step, but AFTER a merge of v0.4 into master.

martinrieder, 2024-06-08:
@kvid @formatc1702
I see so many feature requests and I would even have some more ideas... It seems overwhelming and possibly requires some priorisation or roadmapping.
I suggest that one of you starts an issue for discussion on this topic. Similar to #365, there should be some PR about the next major release already. I do not want to take ownership of it, though. ;)

kvid, 2024-06-08:
I use #365 to collect simple bugfixes to be released soon, and I have chosen to base such PRs on `master` to enable other PRs (not part of the upcoming bugfix release) being based on `dev` in parallel. The first one to be merged into `dev` is #251, but I've been too busy lately to look into that yet.

However, it's normally trivial to rebase a PR branch from `dev` to `master` or vice versa (unless the amount of change conflicts is huge) if priorities are changed. We are all free to argue for which PRs to include in a release, and if you feel e.g. PR #357 is important to release before v0.5, then please describe your reasons for that in a comment. 😃 

Normally, I recommend basing a new PR on `dev`, but if it's a bugfix or other minor change with high priority, then you probably should argue for basing it on `master`.

martinrieder, 2024-06-08:
> I use #365 to collect simple bugfixes to be released soon, and I have chosen to base such PRs on `master` to enable other PRs (not part of the upcoming bugfix release) being based on `dev` in parallel. The next PR to be merged into `dev` is #251, but I've been too busy lately to look into that yet. 

Dito. I get the impression, #251 would definitely demand an 0.5 release branch or even a major step to 1.0...
I would love to see semantic release in action by then, because:

> We are all free to argue for which PRs to include in a release, and if you feel e.g. PR #357 is important to release before a future v0.5, then please describe your reasons for that in a comment. 😃 

That is the reason why [Semantic Versioning](https://semver.org/) states clear rules and leaves out the personal feelings. If we were to follow these, #357 is a feature that would be part of 0.5.

> Normally, I recommend basing a new PR on `dev`, but if it's a bugfix or other minor change with high priority, then you probably should argue for basing it on `master`.

There is another argument for basing it on `master` in the [GitHub Docs about tracking issues and PRs](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue) 
> Note: The special keywords in a pull request description are interpreted when the pull request targets the repository's default branch. However, if the PR's base is any other branch, then these keywords are ignored, no links are created and merging the PR has no effect on the issues._ If you want to link a pull request to an issue using a keyword, the PR must be on the default branch._

kvid, 2024-10-04:
@formatc1702, @amotl, @martinrieder - Please consider my suggested way to merge as described in https://github.com/wireviz/WireViz/pull/408#issuecomment-2381433003 with hight priority, or suggest alternatives.

formatc1702, 2025-01-16:
FYI, I have been invited by @sethhillbrand to give a talk about WireViz in the [Open Hardware and CAD/CAM](https://fosdem.org/2025/schedule/track/hardware/) track of [FOSDEM 2025](https://fosdem.org/2025/) in two weeks, on Feb 2. 🙂

Even though I have not been able to contribute to the project for a while, it will be nice to share this tool and connect with other wiring nerds. If you happen to at the event as well, feel free to ping me (e-mail preferred)!

And again, thank you everyone for keeping the ball rolling... ❤

17o2, 2025-02-09:
Quick heads up: I'm updating my user handle across various platforms*, and this includes GitHub. Sorry if this causes some confusion. This project/repo should not be affected. Some old @ mentions may no longer work, but nothing big should break.

'* from @formatc1702 to @17o2

17o2, 2025-03-08:
The refactored code has been merged into `dev` via #447 , a compacted version of #251. The internal version has been bumped from `v0.5-dev` to `v0.5-dev+refactor`, based on [PEP 440](https://peps.python.org/pep-0440/).

Regarding semantic versioning, IMHO it becomes extremely important once you have a "stable" v1, and need to clearly distinguish between breaking and non-breaking changes. At v0.x, I consider the code to be in flux and reserve the right to fully break backwards compatibility for the sake of improving and maturing the codebase. See the note that has been in the Readme forever ;)

> This is very much a work in progress. Source code, API, syntax and functionality may change wildly at any time.

I look forward to reaching a point where we can talk about a true v1, but that's not today. Inkscape also [took its sweet time](https://en.wikipedia.org/wiki/Inkscape#Release_history) and there's nothing wrong with that.

kvid, 2025-03-14:
@17o2 wrote:
> The refactored code has been merged into `dev` via [#447](https://github.com/wireviz/WireViz/pull/447) , a compacted version of [#251](https://github.com/wireviz/WireViz/pull/251). The internal [...]

Great news!

I've been on travel, and haven't been able to follow up much since I found the solution to #441, verified #443, and created #445 in February.

> Regarding semantic versioning, IMHO it becomes extremely important once you have a "stable" v1, and need to clearly distinguish between breaking and non-breaking changes. At v0.x, I consider the [...]

I agree...

Do you know if anyone recorded your talk at FOSDEM 2025? I would be glad to see it.

17o2, 2025-03-14:
> Do you know if anyone recorded your talk at FOSDEM 2025? I would be glad to see it.

The talks were recorded, and I received a link to review the  video. It had some gaps (technical issues, I guess), so I reported it. I guess it should be released once it's fixed, or once it's clead that the missing bits are gone for good. It should then appear on the [talk page itself](https://fosdem.org/2025/schedule/event/fosdem-2025-4612-wireviz-beautiful-wiring-documentation/).

amotl, 2025-04-22:
> Please consider my suggested way to merge as described in https://github.com/wireviz/WireViz/pull/408#issuecomment-2381433003 with hight priority, or suggest alternatives.

@kvid: Apologies for missing this. @17o2: Nice that you gave a talk at FOSDEM. I am looking forward to watch the video when I get back to better connectivity. @amotl: Note to self: It looks like [wireviz-web](https://github.com/wireviz/wireviz-web) needs a few updates to accompany more recent releases of WireViz.

Thank you everyone for your excellent work on WireViz.