wireviz/WireViz #122: A redesign of wiring blocks. [SUGGESTION]

develroo, 2020-07-23:
So I am new to Wireviz and have nor actually used it yet, but as an Electronic's buff I find the fact that connectors , cables and connections all look the same is somewhat confusing. I suggest that cables (wireing) be in a Round block to differente it from Connectors and Connections. Conceptionally you can imagine it as looking down bundle  and seeing all the indiviual wires in relief. 

Otherwsie it looks quite a handly little tool and I like the ease of configuration. 

Thanks.

aakatz3, 2020-07-24:
The current dev build uses a dashed table outline for the wires, is this sufficient?

formatc1702, 2020-07-24:
If you're only using the code on the `master` branch, you'll be pleased to know that better looking cables are on their way.
Have a look at the [example gallery in the `dev` branch](https://github.com/formatc1702/WireViz/blob/dev/examples/readme.md).
I'm sure it's easy enough to tell cables and connectors apart there :)

> The current dev build uses a dashed table outline for the wires

This is only true for bundles, not cables. Compare examples 02 and 03 in the link above.

kvid, 2020-08-01:
@develroo wrote:
> So I am new to Wireviz and have nor actually used it yet, but as an Electronic's buff I find the fact that connectors , cables and connections all look the same is somewhat confusing. I suggest that cables (wireing) be in a Round block to differente it from Connectors and Connections.

It should be possible to render cables and bundles with rounded corners by adding `STYLE="ROUNDED"` to the outer [HTML](https://graphviz.org/doc/info/shapes.html#html) table in these nodes to obtain a different shape that connectors, but some other changes are also needed to avoid the inner tables going out in the corners and destroy the rounding effect.

To enhance the difference even further, we could also add optional different background colors for different node types. They need to be very light colors to avoid bad contrast together with the black borders and text.

> Conceptionally you can imagine it as looking down bundle and seeing all the indiviual wires in relief.

I'm not quite sure what is meant by this.

formatc1702, 2020-08-01:
Personally, while I'm not opposed to the idea of tweaking the look of the output, to help with clarity, I'm not a big fan of the idaes proposed so far.

- Using rounded corners is possible (the [very first WireViz proof-of-concept](https://twitter.com/formatc1702/status/1263158158656225280/photo/1) used them), the idea of using rounded vs. square edges to distinguish node types feels as aesthetically pleasing as using serif fonts for one type and sans-serif for the other 🤷‍♂️ 
- > To enhance the difference even further, we could also add optional different background colors for different node types.

  Slightly better, but, as an engineer, I feel color (even just shades of gray) should not be relied upon too much in technical drawings. The actual color marks for components are fine, as a bonus, but like I have mentioned before, a technical drawing should work just as well in greyscale, and in strict black & white.
- Maybe I'm biased because I've been looking at WireViz outputs for too long, but are connectors and cables truly hard to tell apart? The fact that the colored wires go straight through the cable nodes is a pretty clear indicator to me...

formatc1702, 2020-08-01:
> > Conceptionally you can imagine it as looking down bundle and seeing all the indiviual wires in relief.
> 
> I'm not quite sure what is meant by this.

Maybe they mean something like this? Like a cross-section?

[image]

develroo, 2020-08-03:
> > > Conceptionally you can imagine it as looking down bundle and seeing all the indiviual wires in relief.
> > 
> > 
> > I'm not quite sure what is meant by this.
> 
> Maybe they mean something like this? Like a cross-section?
> 
> [image]

Yes that was kinda what I was thinking of only obviously an ellipse would be more helpful for bundles of wires than a true circle. But conceptionally you could see then that it is  a different medium. 

Like I say that is how it struck me as a new user but perhaps I am not the only one who thinks that way.

kvid, 2020-08-04:
> > Maybe they mean something like this? Like a cross-section?
> 
> Yes that was kinda what I was thinking of only obviously an ellipse would be more helpful for bundles of wires than a true circle. But conceptionally you could see then that it is a different medium.

If PR #137 is accepted, then it will be possible to embed a cross-section as an image into the cable node, like the example in https://github.com/formatc1702/WireViz/issues/27#issuecomment-668704773. The user must provide the image, though.