wireviz/WireViz #124: [proposed feature] Netlist input/output

aakatz3, 2020-07-24:
It would be nice to have an output and/or input for some form of EDA type netlist, whether in SPICE type format, Protel/Altium format, Kicad format, eagle format, etc.

If people want this feature, I can investigate to see which format is the best.

formatc1702, 2020-07-24:
I hadn't thought of any EDA tool integration (could you describe a specific usecase for this?) but a sort of netlist output has been on my mind.

Simply a table of signals within the harness, and a listing of which wires and connector pins are carrying this signal, would be quite useful when the wiring diagram gets a bit more complex and individual wires harder to follow.

Example output shown as YAML for readability>
```yaml
signals:
  GND:
    - X1: [1, 3]
    - W1: [1, s]
    - X2: [1]
  VCC:
    - X1: [2]
    - W1: [2]
    - X2: [2]
  ```

aakatz3, 2020-07-24:
If you draw schematics for connectors in something like altium or kicad, and wish to migrate or better visualize them in wireviz, you may export a netlist:

* [Kicad Netlist](https://github.com/uvmaero/AERO_2019-2020/blob/master/TSAL_NEW/KiCad/TSAL/TSAL.net)

* [Protel Netlist](https://github.com/formatc1702/WireViz/files/4974389/Cortex.Trace.to.Debug.net.txt)

* [SPICE Netlist](https://github.com/formatc1702/WireViz/files/4974393/SPICE.Netlist.txt)

* [Orcad Netlist.NET.txt](https://github.com/formatc1702/WireViz/files/4974396/Orcad.Netlist.NET.txt)

There are an absolute buttload of formats, but essentially, netlist import would allow wireviz to be able to import and export from schematics from eda tools, or at least spice or tinycad schematics.

Ideally, Wireviz would be able to generate its own format of YML file from the netlist to allow for all of the options it supports to be used.

stevegt, 2020-10-04:
The PySpice netlist parser might be worth experimenting with: https://github.com/FabriceSalvaire/PySpice/blob/319d300fb5f18fc4800d4282c39b2e348d167695/PySpice/Spice/Parser.py#L772