wireviz/WireViz #125: [bug] Shield wires in cable nodes are rendered differently than spline shield wires

kvid, 2020-07-26:
As the spline shield wires are rendered as thin tinned wires with black borders, and the shield wires in cable nodes are rendered as a black single (bottom) border, they don't fit well together, e.g. in `tutorial03.png`:
[image]

formatc1702, 2020-07-26:
true... personally I like displaying shields as thin black wires to match the line in the node, and to make them appear different from a black/grey regular wire. would that be better?

kvid, 2020-07-26:
I will make a change in my PR #126 later today - probably in the evening.

formatc1702, 2020-07-26:
FYI: In the multicolor wires PR, shields were represented as a thin _silver_ wire... My worry with that is that this offers poor contrast, especially when printing the diagram. While using colors is definitely helpful, I'm trying to make all the WireViz features work even when printed in B&W, and to keep it accessible to colorblind users (hence the explicit color names in the wire nodes)

kvid, 2020-07-27:
@formatc1702 wrote:
> true... personally I like displaying shields as thin black wires to match the line in the node, and to make them appear different from a black/grey regular wire. would that be better?

I have now updated PR #126 according to your preference. In addition, it was very easy to also allow a colored shield as the current spline shield wires shown in the description above if the user wants that, by allowing to specify a color in `cable.shield`. See the comparison in the PR description, test it yourself, and let me know what you think.

SnowMB, 2020-07-27:
> In the multicolor wires PR...

I just quickly hacked something because I was messing with the wire representation inside the cable table. It was not meant as a change suggestion. Sorry for the confusion. :sweat_smile:

formatc1702, 2020-07-27:
@SnowMB don't worry, I was talking about the old PR #96. I know your work on stripes is WIP ;)

closed 2020-07-27 by formatc1702