wireviz/WireViz #129: @formatc1702 Improve readability of manufacturer_info_field()

kvid, 2020-07-27:
During the review of #121, I suggested:
@formatc1702 How about this alternative instead?
```py
    if manufacturer:
        return f'{manufacturer}: {mpn}' if mpn else manufacturer
    else:
        return f'MPN: {mpn}' if mpn else None
```
Edit: It's closer to the original, except without a variable and without [parentheses around the return expressions](https://stackoverflow.com/questions/4978567/should-a-return-statement-have-parentheses).

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/121#discussion_r459378362_

formatc1702, 2020-07-27:
Sorry, it was a bit unclear which variant @Tyler-Ward's response was referring to, and I took

> have swapped to that implementation in the latest commits

to mean that the PR already contained his preferred choice.

Personally, I stand behind my proposal, since it separates the "show the MPN cell or not?" logic in the first stage (`if` statement) from the "how to format the cell?" logic in the second stage (`return` statement), and is thus cleaner IMO.
I would argue this is down to personal coding style and not a huge deal-breaker either way :)
 
@Tyler-Ward, please let us know if you'd like to change it, since it's your code contribution and I'm happy to oblige.

Tyler-Ward, 2020-07-28:
> > have swapped to that implementation in the latest commits
> 
> to mean that the PR already contained his preferred choice. Perhaps I misinterpreted this.

That is correct, while the suggestion from @kvid looks cleaner when reading the code. I felt that the sugestion from @formatc1702 made the the fact that it returned either a string or None clearer.

kvid, 2020-07-29:
> That looks like a much tider way to implement that, have swapped to that implementation in the latest commits

> That is correct, while the suggestion from @kvid looks cleaner when reading the code. I felt that the sugestion from @formatc1702 made the the fact that it returned either a string or None clearer.

@Tyler-Ward Thank you for clearing up my misunderstanding. Please excuse me for misinterpreting your comment during the review and thereby creating the unnecessary noise with this issue and PR. English is not my primary language. 🥺

closed 2020-07-29 by kvid