wireviz/WireViz #131: Using multiple color for wire makes ugly output

X-Ryl669, 2020-07-27:
I'm using the dev branch as of today (last commit: b9a4783b6fba882234c3c3e2c74968dcfa38596d )
I'm using the example code as given in my comment in PR#110
I'm getting this output:
[image]

I'd have expected that wires all have the same thickness, but they don't.

formatc1702, 2020-07-27:
Currently, rendered wire thickness it is decided per cable/bundle: if any of the wires of a cable/bundle is multicolored, all the single-colored ones _in that group_ are rendered in the same thickness. This change was introduced in 8cba79334cb3bf689a9a3ace53fa2986bbe5abe7.

I agree that the output above does not look like a user would probably expect, so the check would have to happen on a harness level (is any wire _in the entire harness_ multicolored?).

I welcome a quick fix for this, but I will emphasize that wires should _still_ be rendered in the thin style if no multicolored wires appear in the harness, so as not to make them unnecessarily thick. In the future (see #120) this whole issue should disappear anyway, unless a user specifically needs lengthwise striped wires as opposed to rings.

formatc1702, 2020-07-27:
I've made an adjustment in #132 that addresses the issue. IMHO, this does not look better than the picture posted above.

[image]

X-Ryl669, 2020-07-27:
Yes I know, I've tried to play with it and got very frustrated too by this wavy rendering of graphviz. I prefer the version below than the one above.

closed 2020-07-27 by formatc1702