wireviz/WireViz #138: Allow pin out to be numbered

opolanco23, 2020-07-30:
Hello,

If this is already a feature if you could point me in the right direction that would be awesome. 

I have a 26 pin molex connector where only 6 pins are used, so I would like to show which pins of the 26 are actually used. If just describe the pin out its interpreted as there are only 5 pins if I manually add the pin count then I do not know how to specify which of the 26 are used. 

```
type: automotive
    subtype: male
    pincount: 26
    pinout: [GA-NET+, AMP, RCA+, GA-NET-, RCA-]
    notes: GA-NET(BUS) has 68 ohm 1/2 watt resistor
```

could something like this be used

```
pincount: 26
pinout:
   pin 20: AMP
```

and any pins not mentioned are assumed blank? If there is already something like this then I apologize and please show me an example.

kvid, 2020-07-30:
> Hello,
> 
> If this is already a feature if you could point me in the right direction that would be awesome.
> 
> I have a 26 pin molex connector where only 5 pins are used, so I would like to show which pins of the 26 are actually used. If just describe the pin out its interpreted as there are only 5 pins if I manually add the pin count then I do not know how to specify which of the 26 are used.
> 
> ```
> type: automotive
>     subtype: male
>     pincount: 26
>     pinout: [GA-NET+, AMP, RCA+, GA-NET-, RCA-]
>     notes: GA-NET(BUS) has 68 ohm 1/2 watt resistor
> ```
> 
> could something like this be used
> 
> ```
> pincount: 26
> pinout:
>    pin20: AMP
> ```
> 
> and any pins not mentioned are assumed blank? If there is already something like this then I apologize and please show me an example.

Perhaps not exactly what you want, but you could try using `NC` or `-` for unused pins until your last used pin in the `pinout` and `hide_disconnected_pins` like in this (slightly modified [example 07](https://github.com/formatc1702/WireViz/tree/master/examples#example-07)):
```yaml
# contributed by @elliotmr

connectors:
  X1:
    type: TE 776164-1
    subtype: female
    hide_disconnected_pins: True
    pincount: 35
    pinout: [NC,-,-,-,AA,BB]
    notes: Unconnected pins are not shown

  X2:
    type: D-Sub
    subtype: female
    pincount: 9
    pinout: [NC,B,-,-,-,-,A]
    hide_disconnected_pins: True
    notes: Unconnected pins are not shown

cables:
  C1:
    wirecount: 2
    gauge: 20 AWG
    colors: [YE, GN]
    length: 1

connections:
  -
    - X1: [5,6]
    - C1: [1,2]
    - X2: [7,2] 
```
[image]

kvid, 2020-07-30:
> I have a 26 pin molex connector where only 5 pins are used, so I would like to show which pins of the 26 are actually used. If just describe the pin out its interpreted as there are only 5 pins if I [...]

Based on your your connector info (assuming which 4 pin numbers you are using other than pin 20) and my last alternative in my comment above, I got this:
```yaml
connectors:
  Molex1:
    type: automotive
    subtype: male
    pincount: 26
    pinnumbers: [19,20,21,22,23]
    pinout: [GA-NET+, AMP, RCA+, GA-NET-, RCA-]
    notes: GA-NET(BUS) has 68 ohm 1/2 watt resistor

cables:
  Cable1:
    wirecount: 5
    color_code: DIN

connections:
  -
    - Molex1: [19-23]
    - Cable1: [1-5]
```
[image]

formatc1702, 2020-07-30:
@kvid what branch are you using to generate the example above? `pinout` should be `pinlabels` and `pinnumbers` should be `pins`?

kvid, 2020-07-30:
> @kvid what branch are you using to generate the example above? `pinout` should be `pinlabels` and `pinnumbers` should be `pins`?

I did use `master` because @opolanco23 used `pinout` in his question, and therefore was not using the `dev` branch.

> It is important to add that this will only work in the `dev` branch currently, not on `master`.

Are you sure? I just did that without any errors.

formatc1702, 2020-07-30:
Huh, I must have misremembered when this feature was introduced. You're right.

opolanco23, 2020-07-31:
@kvid thanks for the info, what is in the master branch now will work. Feel free to close this

closed 2020-07-31 by formatc1702