wireviz/WireViz #142: [feature] Enable including components in the harness

kvid, 2020-07-31:
Sometimes, a few simple components are integrated into the harness, like e.g. a LED indicator or switch mounted at a connector housing, a resistor at the end of a bus line, diodes for protecting against reversed polarity, etc. I suggest adding such a component as a connector, except that it can be marked as a Component instead of a Connector in the BOM entries.

In addition, it would be nice to have the option to connect the pins of such a component directly to a splice (simple connector) or connector pin without cable/wires between when that is how to do it in real life. It might be implemented similar to the mockup in https://github.com/formatc1702/WireViz/issues/134#issuecomment-665877504 where ferrules (simple connectors) are connected directly to another connector.

Are there other use cases where something like this might be useful?

formatc1702, 2020-08-02:
This sounds extremely useful, but will require some rewriting of the input checking logic, since so far it has been a strict CABLE-CONNECTOR-CABLE-CONNECTOR-.... sequence.

It's a good task to tackle for v0.3!

kvid, 2020-08-09:
> This sounds extremely useful, but will require some rewriting of the input checking logic, since so far it has been a strict CABLE-CONNECTOR-CABLE-CONNECTOR-.... sequence.

I have suggested a possible input syntax in https://github.com/formatc1702/WireViz/issues/134#issuecomment-671000507 that doesn't break the alternating connector-cable-sequence, except accepting a direct connection as an alternative to a cable in the sequence , and that also can support mating wire connectors within the harness, as in #148.

formatc1702, 2020-10-21:
Naïve question:
Is there a way to unify connectors and these components somehow in the YAML?
I understand that currently, the BOM entry auto-appends the "Connector" prefix, which is unwanted in this case, but easily fixed.

Alternatives:
- Rename `connectors` to `components`, and adding a `category: connector` parameter, which would not necessarily have to be rendered inside the node, but could be prepended in the BOM.
  - The problem is that, technically, cables and wires are components too, so the semantic distinction kind of falls apart.
  - Most components are likely to be connectors in the majority of use cases, so this adds a lot of unnecessary lines to the YAML if you need to specify this every time.
- Keep connectors and components separate, just like connectors and cables are separate.
  - BOM generation might get uglier, since now one more dictionary need to be parsed...
  - As mentioned above, a better semantic distinction would be helpful. 
- Simply omit the "Connector" in the BOM entry, and treat these components as connectors.
  - IMHO, LEDs or switches within a wire/harness kind of qualify as "connectors" too, so it's not completely far fetched to list them under the `connectors` list. But it's not the most elegant solution either.

Any other ideas?

formatc1702, 2020-11-12:
This may be only partially related, but this is how I implemented including a little level shifter PCB in one of our wiring diagrams at work (using the `feature/mate+autogenerate` branch for the arrows, and the dashed lines representing the "internal routing" within the PCB.

[image]

My vision is to be able to group connectors using the GraphViz `cluster` feature to more clearly show that both `*_M` connectors are part of the same component.