wireviz/WireViz #143: [meta] Example contributions

formatc1702, 2020-08-01:
If you have used WireViz for a project, and want to showcase your work, feel free to post it here!
We may add it to the example gallery as well, provided you allow it :)

formatc1702, 2020-08-01:
@ericbhanson
I want to thank you for your post in #120! I thought I'd start this thread and get back to you here so as not to derail that thread. It's amazing seeing WireViz being used for more complex harnesses. This is good for pushing it a bit and seeing how it performs for more demanding applications. It really highlights the need for a netlist, for example ;)

Would you mind if we added the harness above to the example gallery?

Also, there are things you could simplify in your YAML while retaining the same output!

kvid, 2020-08-02:
- I'm not sure if this suggestion belongs here, or in a separate issue. I can move this comment if needed.

- I have a script that generates an example harness for each series of color codes. Maybe that would fit into a new `docs/colors` subfolder and linked from the `docs/syntax.md` in PR #111, or somewhere else? If you want to include these, I can create a PR to integrate my script into [build_examples.py](https://github.com/formatc1702/WireViz/blob/dev/src/wireviz/build_examples.py).

- Three of my examples:
  - [YAML source](https://github.com/formatc1702/WireViz/files/5011593/Color.codes.yml.txt) - [Color codes image](https://user-images.githubusercontent.com/12657033/89114598-92acfe00-d47e-11ea-8878-c6e977a99080.png)
  - [YAML source](https://github.com/formatc1702/WireViz/files/5011619/DIN.codes.yml.txt) - [DIN codes image](https://user-images.githubusercontent.com/12657033/89115051-b161c380-d483-11ea-9641-ab0e2cf1132b.png)
  - [YAML source](https://github.com/formatc1702/WireViz/files/5011595/IEC.codes.yml.txt) - [IEC codes image](https://user-images.githubusercontent.com/12657033/89114614-c12ad900-d47e-11ea-9834-a702558515a4.png)

- The TEL and TELALT codes examples depend on a fix for #144.

ericbhanson, 2020-08-02:
@formatc1702 it's my pleasure - I'm glad I could help move the project forward! I would be honored if my harness were added to the gallery. 

Thank you for the notes about how to DRY the YAML. I was actually using templates initially, but I'm now getting an error when doing so, so I've removed them. Here's a example:

When attempting to process the following YAML file:
```connectors:
  Bullet: &bullet
    subtype: male
    pincount: 1
  GR-SW-M:
    <<: *bullet
    notes: Green Shockwave Module
```

I get the following stack trace:
```Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/wireviz", line 11, in <module>
    load_entry_point('wireviz', 'console_scripts', 'wireviz')()
  File "/Users/eric/Downloads/WireViz/src/wireviz/wireviz.py", line 248, in main
    parse(yaml_input, file_out=file_out)
  File "/Users/eric/Downloads/WireViz/src/wireviz/wireviz.py", line 178, in parse
    harness.output(filename=file_out, fmt=('png', 'svg'), view=False)
  File "/Users/eric/Downloads/WireViz/src/wireviz/Harness.py", line 430, in output
    graph = self.create_graph()
  File "/Users/eric/Downloads/WireViz/src/wireviz/Harness.py", line 392, in create_graph
    g = graph_from_dot_data(dot.create_dot(prog="dot", f="dot").decode('utf-8'))[0]
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pydot.py", line 1722, in new_method
    return self.create(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pydot.py", line 1945, in create
    assert process.returncode == 0, process.returncode
AssertionError: 1
```

As far as bundling wires by color: that was an organizational choice to make it easier to make sure I had everything wired up properly. To be honest, I'm a little confused by the purpose of bundles; what are they intended to do?

formatc1702, 2020-08-02:
As for the templates, I'll have to look into it in some more detail.

> As far as bundling wires by color: that was an organizational choice to make it easier to make sure I had everything wired up properly. To be honest, I'm a little confused by the purpose of bundles; what are they intended to do?

You can picture bundles as individual wires being routed together physically (as if zip-tied into a literal bundle).
If you had one bundle of wires going from your fusebox to your handlebars, and another one going from your fusebox to the engine, and another to the rear of the bike, you could organize them as such.
Currently, all wires in a bundle are assumed/forced to be the same length, which makes sense since they will be routed together... but this restriction will likely be loosened in the future.
A big difference is that _cables_ are listed as a single BOM entry (one 4-wire cable of length X) where as bundles list every wire individually (four wires of length X).

formatc1702, 2020-08-02:
@kvid your color code test script sounds like a good tool for development, but including all the examples in the repo is probably overkill... maybe we could add a new directory for these kinds of tools?

ericbhanson, 2020-08-04:
Thanks @formatc1702 - I'm going to recreate the configuration with proper use of bundles in mind.