wireviz/WireViz #144: [bug] TEL and TELALT contains some unknown color codes

kvid, 2020-08-02:
- Both `COLOR_CODES['TEL']` and `COLOR_CODES['TELALT']` contain 10 entries each with the color code `'YW'` that I guess should have been `'YE'` for yellow. Is that correct, or is it a different yellow color that should be inserted into the `_color_hex`, `_color_full`, and `_color_ger` tables? The error message `Unknown color specified` is repeated 20 times while running each test case below.

- TEL test case:
```yaml
# Generated by build_colors.py
connectors:
  TEL:
    show_pincount: false
    pinlabels: [blue/white, white/blue, orange/white, white/orange, green/white, white/green, brown/white, white/brown, slate/white, white/slate, blue/red, red/blue, orange/red, red/orange, green/red, red/green, brown/red, red/brown, slate/red, red/slate, blue/black, black/blue, orange/black, black/orange, green/black, black/green, brown/black, black/brown, slate/black, black/slate, blue/yellow, yellow/blue, orange/yellow, yellow/orange, green/yellow, yellow/green, brown/yellow, yellow/brown, slate/yellow, yellow/slate, blue/violet, violet/blue, orange/violet, violet/orange, green/violet, violet/green, brown/violet, violet/brown, slate/violet, violet/slate]

cables:
  TEL codes:
    wirecount: 50
    color_code: TEL

connections:
  -
    - TEL codes: [1-50]
    - TEL: [1-50] 
```

- TELALT test case:
```yaml
# Generated by build_colors.py
connectors:
  TELALT:
    show_pincount: false
    pinlabels: [white/blue, blue, white/orange, orange, white/green, green, white/brown, brown, white/slate, slate, red/blue, blue/red, red/orange, orange/red, red/green, green/red, red/brown, brown/red, red/slate, slate/red, black/blue, blue/black, black/orange, orange/black, black/green, green/black, black/brown, brown/black, black/slate, slate/black, yellow/blue, blue/yellow, yellow/orange, orange/yellow, yellow/green, green/yellow, yellow/brown, brown/yellow, yellow/slate, slate/yellow, violet/blue, blue/violet, violet/orange, orange/violet, violet/green, green/violet, violet/brown, brown/violet, violet/slate, slate/violet]

cables:
  TELALT codes:
    wirecount: 50
    color_code: TELALT

connections:
  -
    - TELALT codes: [1-50]
    - TELALT: [1-50] 
```

- Suggested fix:
```diff
----------------- src/wireviz/wv_colors.py ---------------------------
index 4157543..eff6672 100644
@@ -13,11 +13,11 @@ COLOR_CODES = {
     'BW': ['BK', 'WH'],
     'TEL': ['BUWH', 'WHBU', 'OGWH', 'WHOG', 'GNWH', 'WHGN', 'BNWH', 'WHBN', 'SLWH', 'WHSL', 'BURD', 'RDBU', 'OGRD',
             'RDOG', 'GNRD', 'RDGN', 'BNRD', 'RDBN', 'SLRD', 'RDSL', 'BUBK', 'BKBU', 'OGBK', 'BKOG', 'GNBK', 'BKGN',
-            'BNBK', 'BKBN', 'SLBK', 'BKSL', 'BUYW', 'YWBU', 'OGYW', 'YWOG', 'GNYW', 'YWGN', 'BNYW', 'YWBN', 'SLYW',
-            'YWSL', 'BUVT', 'VTBU', 'OGVT', 'VTOG', 'GNVT', 'VTGN', 'BNVT', 'VTBN', 'SLVT', 'VTSL'],
+            'BNBK', 'BKBN', 'SLBK', 'BKSL', 'BUYE', 'YEBU', 'OGYE', 'YEOG', 'GNYE', 'YEGN', 'BNYE', 'YEBN', 'SLYE',
+            'YESL', 'BUVT', 'VTBU', 'OGVT', 'VTOG', 'GNVT', 'VTGN', 'BNVT', 'VTBN', 'SLVT', 'VTSL'],
     'TELALT': ['WHBU', 'BU', 'WHOG', 'OG', 'WHGN', 'GN', 'WHBN', 'BN', 'WHSL', 'SL', 'RDBU', 'BURD', 'RDOG', 'OGRD',
                'RDGN', 'GNRD', 'RDBN', 'BNRD', 'RDSL', 'SLRD', 'BKBU', 'BUBK', 'BKOG', 'OGBK', 'BKGN', 'GNBK', 'BKBN',
-               'BNBK', 'BKSL', 'SLBK', 'YWBU', 'BUYW', 'YWOG', 'OGYW', 'YWGN', 'GNYW', 'YWBN', 'BNYW', 'YWSL', 'SLYW',
+               'BNBK', 'BKSL', 'SLBK', 'YEBU', 'BUYE', 'YEOG', 'OGYE', 'YEGN', 'GNYE', 'YEBN', 'BNYE', 'YESL', 'SLYE',
                'VTBU', 'BUVT', 'VTOG', 'OGVT', 'VTGN', 'GNVT', 'VTBN', 'BNVT', 'VTSL', 'SLVT'],
     'T568A': ['WHGN', 'GN', 'WHOG', 'BU', 'WHBU', 'OG', 'WHBN', 'BN'],
     'T568B': ['WHOG', 'OG', 'WHGN', 'BU', 'WHBU', 'GN', 'WHBN', 'BN'],
```

formatc1702, 2020-08-02:
It clearly is meant to be yellow. Thanks for catching this.

kvid, 2020-08-02:
> It clearly is meant to be yellow. Thanks for catching this. Can you submit a quick PR?

I'm busy now, but in a few hours I should be able to do it.

closed 2020-08-12 by formatc1702