wireviz/WireViz #148: Connector to Connector Mating

jtroeth1, 2020-08-06:
Hi,

Is there currently a way to mate two connectors together without a wire in between?
Eg a cable mount male connector mounting to a panel mount female connector.

In the attached image I would like to join X2 & X3 but X3 just loops onto itself.
[image]

kvid, 2020-08-07:
> Is there currently a way to mate two connectors together without a wire in between?

Currently, this is not possible, but similar requests have been discussed. Your request seems to be related to (maybe a generalization of) #134 and a solution is also relevant to #142. See if the mockup in https://github.com/formatc1702/WireViz/issues/134#issuecomment-665877504 looks like something useful for you, with the dashed arrows between your X2 and X3?

formatc1702, 2020-10-21:
Closing to consolidate with #134, will generalize title there to include this use case.

closed 2020-10-21 by formatc1702