wireviz/WireViz #152: [doc] Improve Contribution Guidelines

kvid, 2020-08-13:
I suggest adding some items in the [Contribution Guidelines](https://github.com/formatc1702/WireViz/blob/dev/CONTRIBUTING.md):
- Recommend creating a separate branch for each feature/bugfix that is pushed to your fork. (Unique branch name for each PR.)
- Recommend the advises from [How to Write a Git Commit Message](https://chris.beams.io/posts/git-commit/).
- If the `dev` branch has deviated from the PR branch before merge-time, recommend rebasing the PR branch on top of the new `dev` commits and resolve any conflicts before merge.
- If the PR branch contains branch merges or a high number of intermediate commits, recommend squashing some of them together using `git rebase -i` to simplify the commit history before merge.
- Maybe include the essential git commands?

formatc1702, 2020-08-13:
I've added some of your points in the latest commit to #111.

> Maybe include the essential git commands?

I'd rather not write a git tutorial... especially since I am not a CLI git user myself.
And interactive rebasing is an art unto itself.
If you have a good source, we can link to it, though!

kvid, 2020-08-15:
> I've added some of your points in the latest commit to #111.

Thank's

> 
> > Maybe include the essential git commands?
> 
> I'd rather not write a git tutorial... especially since I am not a CLI git user myself.
> And interactive rebasing is an art unto itself.
> If you have a good source, we can link to it, though!

I've no such link suggestions at the moment

closed 2020-08-15 by kvid