wireviz/WireViz #155: [feature]Cable glands support?

martonmiklos, 2020-08-15:
Hi!

Many thanks for sharing this project, it really rocks!

I am working on creating a documentation for a cable where I would need to add a cable gland to the cable during the cable assembly. 

I have seen the discussion in #31 about similar thing and it might be useful to bring this up for further discussion because it could  be useful for others as well.

kvid, 2020-08-19:
- With the current (master branch) version, you can add notes to the cable that describes the glands.
- With the current [dev branch](https://github.com/formatc1702/WireViz/tree/dev) (that probably soon will become a new release), you can also add custom BOM entries. This feature is currently only described in the [syntax documentation draft](https://github.com/formatc1702/WireViz/blob/feature/syntax-description/docs/syntax.md).
- With the same new release, it probably also will be possible to include a custom image inside the cable node, in case that might help you visualize cable related properties, e.g. a gland.
- Any implementation of #31 is not yet suggested. Feel free to contribute with your use case and ideas also in that discussion.

martonmiklos, 2020-08-19:
Well I have to think more about how would it be the most elegant/straightforward.

Adding glands as a note could be problematic since they have installation direction, they could have an installation position in a given distance between connectors etc. 

Graphic wise I would imagine something like this:
[image]
(Basically a  vector drawing of a gland which diameter would fit to the wires.)

izyb, 2024-06-07:
Moving request here from #31:

I am trying to spec a cable that has a ferrite attached to one end of it.  Here is my application:

[image]

It would be nice to be able to support the following:
* A part that can fit around a cable, relative to the wire definition block (in the image above, the ferrite is closer to one side of the cable, so I would like to position it on the right side of the cable definition block for clarity to the CM).
* Include BOM + additional BOM items section for the gland/section, just like connectors, which allows me to account for both the ferrite and heatshrink above (other similar applications could use glue or some other attachment method).
* Specify **position along the cable** where the added part will be attached.
* Notes section

martinrieder, 2024-06-08:
Simply referring to #222 as per  https://github.com/wireviz/WireViz/issues/31#issuecomment-2155337263. Details to follow...