wireviz/WireViz #156: [internal] Using the Optional[] keyword in dataclasses

formatc1702, 2020-08-16:
Currently, some fields of the classes in `DataClasses.py` use the `Optional[]` keyword, and some don't.
```python
class Connector:
    ...
    color: Optional[str] = None
    show_name: bool = None
```

I am unsure if there is a difference in behavior between both; or if the ` = None` is needed at all.
Boolean values should probably have a default value of `False` anyway, instead of `None`.

Perhaps someone can chime in and suggest which variant is recommended; then I can create a PR to make the syntax more consistent. Thanks!

kvid, 2020-08-16:
> ...; or if the ` = None` is needed at all.
> Boolean values should probably have a default value of `False` anyway, instead of `None`.

When ` = None` is used, then it's possible to differ between setting the attribute to false and not specifying it in the YAML file. For some attributes this difference is important, e.g. when a default value depends on other factors, like for `show_pincount`:
```py
        if self.show_pincount is None:
            self.show_pincount = self.style != 'simple' # hide pincount for simple (1 pin) connectors by default
```

kvid, 2020-08-20:
> Currently, some fields of the classes in `DataClasses.py` use the `Optional[]` keyword, and some don't.
>
> ```python
> class Connector:
>     ...
>     color: Optional[str] = None
>     show_name: bool = None [...]

There are two different concepts:
- A field can have a default value. The generated  `__init__()` method will then get an optional argument for this field, and no exception is raised when a value for this field is missing in the YAML input.
- A field can have an optional type, i.e. `None` is allowed as a value in addition to the annotated type.

`Optional[X]` is used in the latter case when the type is optional, independently of whether it has a default value or not, but when a default value is `None`, then the type also is optional.

Excerpt from the [doc](https://docs.python.org/3/library/typing.html#typing.Optional) below:

Optional type.

`Optional[X]` is equivalent to `Union[X, None]`.

Note that this is not the same concept as an optional argument, which is one that has a default. An optional argument with a default does not require the `Optional` qualifier on its type annotation just because it is optional. For example:
```
def foo(arg: int = 0) -> None:
    ...
```
On the other hand, if an explicit value of `None` is allowed, the use of `Optional` is appropriate, whether the argument is optional or not. For example:
```
def foo(arg: Optional[int] = None) -> None:
    ...
```

kvid, 2020-09-05:
Should we start using type annotations in other code than `DataClasses.py` as well?

formatc1702, 2020-11-01:
I guess I can leave this issue open even after closing #163 in response to @kvid's last comment:

> Should we start using type annotations in other code than `DataClasses.py` as well?

because yes, I believe it's worth it including more type hints.

PR #163 (2020-11-01): Update the types of dataclass attributes according to usage
Fixes #156
commit 3f091bb4 (2020-11-01): Update the types of dataclass attributes according to usage

Fixes #156