wireviz/WireViz #160: Pin case sensitivity.

zombielinux, 2020-08-28:
I'm looking at implementing this but I use a lot of mil-circ connectors. 

As such, they typically run A-Z, then a-z, then I think aa-zz.

As it stands, defining the template:
```
   - &MS3126F18-32P
     type: MS3126F18-32P
     pins: [A, B, C, D, E, F, G, H, J, K, L, M, N, P, R, S, T, U, V, W, X, Y, Z, a, b, c, d, e, f, g, h, j]
```
And then using pin "a" actually connects pin "A". In my case, these are VERY different pins. This is with the latest dev branch.

kvid, 2020-08-29:
Thank you for reporting this bug. It seems WireViz is case sensitive about identifiers, but unfortunately Graphviz is not. This means a similar problem occurs when a connector/cable name is not case insensitive unique.

I suggest keeping WireViz case sensitive to allow use cases like yours, but appending a unique numeric suffix to those identifiers that are not case insensitive unique (or to all identifiers to simplify the implementation) when writing the identifiers to the `.gv` file (e.g. A_0 and a_1 in your case), except when used as pure text output (i.e. hiding these identifier suffixes for the user).

formatc1702, 2020-10-20:
Here's a quick test that displays the observed behavior:
```
connectors:
  X1:
    pins: [AA, Aa, aA, aa]
  X2:
    pins: [ZZ, Zz, zZ, zz]

cables:
  W1:
    wirecount: 1

connections:
  -
    - X1: AA
    - W1: 1
    - X2: zz
    # expected result: link AA to zz
    # actual result:   link aa to zz
```

formatc1702, 2020-10-20:
> I suggest keeping WireViz case sensitive to allow use cases like yours, and use pin index instead of pin name in all connector port identifiers to ensure their uniqueness.

+1 👍

formatc1702, 2020-11-11:
The safest approach is to use an internally assigned, numerical, auto-incremented ID to every pin (analogously, assign a similar ID to every wire in a cable to iterate over) and map the user-specified pin ID to this internal one.

In the typical case where a user does not specify a connector's `pins` directly, and they are thus auto-numbered, the internal ID and the "user-visible" ID used when defining a connection will simply be the same.

In the above mentioned case, Python/WireViz will case-sensitively match the user-provided `pins` to their respecive unambiguous numerical IDs.

This also solves a semi-related issue:
When using `pins` with unusual characters (`<`,`>` and others), GraphViz is unable to draw the edges because the port names it uses to attach the edge to a node currently contain the `pins` attribute, causing issues with GraphViz' internal HTML.

PR #229 (2021-08-23): Use unique index for connector pin ports
Closes #160.

Use unique numerical index instead of string (which might cause issues when using pin names that are not unique when ignoring upper/lower case).

This does not address the issue of having special HTML characters as part of pin names, labels, etc. [as mentioned in #160](https://github.com/formatc1702/WireViz/issues/160#issuecomment-725544366) and now listed as #230; there should be a separate PR dealing with escaping these entities in the GraphViz HTML (e.g. `>` to `&gt;`).
commit e212fc90 (2021-08-23): Use unique index for connector pin ports (#229)

Closes #160.
closed 2021-08-23 by formatc1702