wireviz/WireViz #166: [RFC]Merge the additional BoM item lines if having matching name with a connector in the cable

martonmiklos, 2020-09-11:
Hi folks!

Recently I created a cable documentation for a contract manufacturer where I faced with the following issue:
One end of the cable needed to be equipped with crimps only, without connector installed and they need to bundle one connector to each cable assembly, so it adds that the connector should be added to the BoM.
The other end of the cable is equipped with the same connector, so one item from that connector is already added to the BoM.
It is straightforward to add this extra connector manually to the additional BoM items however the lines does not get merged to the connector in the cable even if they have the exactly same name:
[image]

I could imagine two solutions for this problem: 
A) Sixup the BoM generation to merge the identical lines
B) Introduce some syntax to being able to referring to connectors from the additional BoM items

I support the A solution due to it's simplicity. B variant I think just would add some more complexity which might not make a lot of sense.

Tyler-Ward, 2020-09-17:
Adding deduplication for the bom lines sounds sensible. I am working on adding the option for specifying additional components for a connector or cable in #115 which will add even more ways to creater duplicate BOM entries.

From a code standpoint there are several seperate deduplication steps in the bom generation code, The current implementation of the additional components sections add more as well, populating all items into a list and then deduplicating at the end might simplify this code as well.

Tyler-Ward, 2020-10-22:
Deduplication of BOM items was implemented as part of #115 which has now been merged into dev, this should cover your use case.

The following fields will need to match to be deduplicated.
* Item
* Unit
* P/N
* Manufacturer
* Mpn

kvid, 2020-10-31:
@martonmiklos are you able to test the YAML input that caused the problem you describe using the current `dev` branch and check if your problem now is solved?

martonmiklos, 2020-11-06:
@kvid Thanks for the heads up, the dev does the merging correctly!

closed 2020-11-06 by martonmiklos