wireviz/WireViz #167: [fix] Minor changes of command line options

kvid, 2020-09-13:
1. ```
usage: wireviz.py [-h] [-o OUTPUT] [--generate-bom] [--prepend-file YAML_FILE]
                  YAML_FILE
```
It seems, however, that the current implementation ignores `--generate-bom`, and I suggest we comment out that `parser.add_argument()` code line until it gets implemented to avoid any confusion.

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/111#discussion_r487470761_

2. I also suggest adding an optional `-b BRANCH` argument to `build_examples.py` to enable comparing to and restoring from any branch or commit. That will help when e.g. reversing an unfortunate commit of generated files.

kvid, 2020-10-10:
I also suggest including a `-V` command line option (and `--version` as an alias) to both `wireviz.py` and `build_examples.py` that output version information. In #157 `-v` was suggested, but that is often used for verbose mode, and both `python` and `dot` use `-V` for version information.

I suggest specifying the version number in `__init__.py` as done e.g. in [the howdoi project](https://github.com/gleitz/howdoi/blob/master/howdoi/__init__.py). It is then important to not include code that import anything in the same file to avoid any issues when the file is also imported in `setup.py`.

I know you have argued against _feature creep_ in https://github.com/formatc1702/WireViz/issues/101#issuecomment-703774009, but I still feel this feature is important to include in v0.2 to later make it easy to identify the version in use when more than one version has been released. I'm sorry I didn't suggest this much earlier.

formatc1702, 2020-10-10:
I 100% agree it's a good time to add version output now, so I am willing to include this in v0.2!

I don't mind changing `compare` to `diff` either.

Commenting out unused lines is perfectly reasonable.

Regarding the `-b BRANCH` option... feel free to implement it if you wish, you are probably the one who is using this feature most, and I don't mind having that option. I am not sure how hard it would be to implement/test this, so perhaps it is wise to do this as a separate PR that can be added at a later date in v >= 0.3. Up to you.

closed 2020-10-11 by formatc1702