wireviz/WireViz #170: Format: "png" not recognized

edinc90, 2020-09-28:
I installed Wireviz using pip on Windows 10. When I run wireviz wire2.yml, I get the following error.

```
C:\Windows\System32>wireviz "c:\Users\username\Desktop\WireViz\wire2.yml"
Format: "png" not recognized. Use one of:
Traceback (most recent call last):
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\username\AppData\Local\Programs\Python\Python38-32\Scripts\wireviz.exe\__main__.py", line 7, in <module>
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\wireviz\wireviz.py", line 235, in main
    parse(yaml_input, file_out=file_out, generate_bom=args.generate_bom)
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\wireviz\wireviz.py", line 182, in parse
    harness.output(filename=file_out, fmt=('png', 'svg'), gen_bom=generate_bom, view=False)
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\wireviz\Harness.py", line 219, in output
    graph.render(filename=filename, directory=directory, view=view, cleanup=cleanup)
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\graphviz\files.py", line 207, in render
    rendered = backend.render(self._engine, format, filepath,
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\graphviz\backend.py", line 221, in render
    run(cmd, capture_output=True, cwd=cwd, check=True, quiet=quiet)
  File "c:\users\username\appdata\local\programs\python\python38-32\lib\site-packages\graphviz\backend.py", line 183, in run
    raise CalledProcessError(proc.returncode, cmd,
graphviz.backend.CalledProcessError: Command '['dot', '-Tpng', '-O', 'wire2']' returned non-zero exit status 1. [stderr: b'Format: "png" not recognized. Use one of:\r\n']

```

kvid, 2020-09-28:
Thank you for reporting this error. What version of Graphviz do you have? Run `dot -V` i a command shell to find out. In general, I also recommend including your YAML input file unless (preferably) you can demonstrate the same error using one of the example files. If your YAML input is large or complex, please try to simplify it and check if you still get the same error.

George-PR, 2020-09-28:
I'm seeing the same issue. Just installed fresh wireviz and graphviz.
`dot - graphviz version 2.44.1 (20200629.0846)`

Same issue with x86 and x64 releases.

Uninstalling graphviz 2.44 and installing graphviz 2.38 fixed the issue for me.
https://graphviz.gitlab.io/_pages/Download/windows/graphviz-2.38.msi

edinc90, 2020-09-28:
Well now I have another problem. I uninstalled Graphviz 2.44.1 and installed 2.38, but when I type dot -V, it errors because it's still looking for dot.exe in the 2.44.1 folder. I've changed the PATH in Windows, but it doesn't seem to change anything.

kvid, 2020-09-28:
It seems the [installation procedure at Windows has changed for Graphviz version 2.44](https://forum.graphviz.org/t/new-simplified-installation-procedure-on-windows/224). Please check for any error message when running `dot -c` as Administrator. If this command fails, then the plugins are not configured correctly and the image formats might not be recognized.

edinc90, 2020-09-28:
That fixed it. I reinstalled 2.44.1 then ran dot -c in an administrator command window and it works now. Thanks.

closed 2020-09-28 by edinc90

formatc1702, 2020-12-06:
If this is a common issue, we should add a hint about it in the readme, next to the Ubuntu 18.04 LTS warning maybe?