wireviz/WireViz #172: [install] How to use requirements.txt correctly

kvid, 2020-10-10:
The current contents of `requirements.txt` have some issues.

@formatc1702 wrote:
> Please add PIL / pillow to both `requirements.txt` and `setup.py` (under `install_requires=[`).
> [...]

I have implemented your request, but after testing it, I have some questions:

Why do you want the run-time dependencies both in `requirements.txt` and in `setup.py` when the latter seems to be enough?
- The first line of `requirements.txt` contains `.`, and that means, AFAIK, that `setup.py` is executed to find dependencies specified there, and then adding the same dependencies again from the remaining lines of `requirements.txt` will not change anything.
- The last line of `requirements.txt` contains `setuptools`, but if that is not already installed before running `pip install -r requirements.txt`, then it will fail running `setup.py` when processing the first line containing `.`.
- Moving `setuptools` above the `.` in `requirements.txt` will still fail if `setuptools` is not already installed, probably because all dependencies are processed before running any installation.
- Why do you really need `requirements.txt` in the repo at all? it does not seem to be used by the project installation instructions. Maybe the installation instructions should have an initial `pip install setuptools` to make sure that is installed before running `setup.py`?

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/153#issuecomment-706470565_

kvid, 2020-10-10:
@formatc1702 wrote in https://github.com/formatc1702/WireViz/pull/153#issuecomment-706522281:
> TBH, I am not too familiar with the processes behind `pip`, `setuptools`, etc.
> I have seen a lot of projects using both, and [this post](https://stackoverflow.com/questions/43658870/requirements-txt-vs-setup-py) for example, seems to say that, while there is a way to reduce the duplicate references, if is not unreasonable to keep both files. That's why, for now, I'd like to go the safe route and keep both, too.
> In the future, we could open a new issue/PR to optimize the setup process, if it is deemed necessary...

I read the same post yesterday that you link to, and I concluded there are different opinions about the reason behind `requirements.txt`:
 - Some argue that it's used for installing everything needed for module development without installing the module, but then it doesn't make sense to include `.` (that does install this module).
 - Some argue to include just `.` and nothing else to let `setup.py` do the installation job, but then why store and use the `requirements.txt` at all?
 - Some argue to store the output from `pip freeze` into `requirements.txt` as a documentation on the exact development environment, but in this project we cannot expect all developers to use the exact same set of versions.
 - In any case it doesn't make sense to include both `.` and `setuptools` in the same `requirements.txt`, as far as I can see.

I will update both files when that is your request, even though I don't like this double specification of the exact same information. A new issue is a good idea, and I will probably create that later today.

_Originally posted by @kvid in https://github.com/formatc1702/WireViz/pull/153#issuecomment-706541848_

@formatc1702 anwered in https://github.com/formatc1702/WireViz/pull/153#issuecomment-706557661:

Yes, probably there are two of these views in our source code right now, clashing.

I think the first argument, of removing `.` from `requirements.txt` is the most reasonable, so let's do that for now. Since I am unsure of any other ramifications of emptying/removing `requirements.txt` at this point, let's keep the duplicate dependency info for now and discuss in a new issue.
Thanks!

_Originally posted by @formatc1702 in https://github.com/formatc1702/WireViz/pull/153#issuecomment-706557661_

kvid, 2020-10-12:
The redundancy part of this issue was also discussed by @gstein and @n42 in PR #58.
See also [the commit history](https://github.com/formatc1702/WireViz/commits/dev/requirements.txt):
1. @n42 created the initial contents - a single `.` in PR #9 (which was included in release 0.1)
2. @gstein appended `graphviz` and `pyyaml` in PR #58 
3. @aakatz3 appended `setuptools` in PR #96 
4. @kvid insert `pillow` in #153 (not yet pushed)

I hope these contributors (and others) can comment my conclusions in this issue.

I vote for removing the conflict between `.` and `setuptools` before releasing v0.2, and unless anyone argue against it, removing `.` from `requirements.txt` as concluded by @formatc1702 is probably what will happen.

I would also like an answer to the question: Who needs installing everything needed for module development without installing the wireviz module itself? Why not just install the module using `pip install -e .`?

gstein, 2020-10-12:
I view `requirements.txt` as a simple "what's needed?" .. the inclusion of "." is a bit strange, and I didn't even know that was A Thing.

IMO, I have no voice here, as I have little contribution. I like the clarity of `requirements.txt`, but defer to y'all on its contents.

Thank you for asking!

commit 71a1c52e (2021-03-20): Remove `.` from requirements.txt

Closes #172.