wireviz/WireViz #175: SVG Table borders are exeeded.

norator42, 2020-10-13:
If a field is too long, the table border is exceeded. This error happens only in SVG output, not in PNG output. My temporary solution is to convert the GV file directly into a PDF with: "dot -Tpdf example.gv -o example.pdf" to create a correct vector graphic. Is it possible to fix the SVG output? I think this is a graphviz error - so can you add a direct PDF output for propper vectorized graphics?
[image]
With code:
```
connectors:
  X1:
    pinlabels: [One, Two, Three]
    manufacturer: Manu1
    mpn: TODO
    type: Power
    subtype: Male
    notes: | 
      Here is a very long note text with very much information about the cables.
  X2:
    pinlabels: [One, Two]
    manufacturer: Here is a long Manufacturer
    mpn: 123456789ABCDEF
    type: Power
    subtype: Male
  X3:
    style: simple
    manufacturer: Manufacturer
    mpn: 1234
    type: Ring Terminal
    
cables:
  Power Cable:
    manufacturer: Here is a long Manufacturer
    mpn: 123456789ABCDEF
    wirecount: 2
    length: 5
    gauge: 22 AWG
    colors: [RD, BK]

  Cable 1:
    manufacturer: Here is a long Manufacturer
    mpn: 123456789ABCDEF
    wirecount: 1
    length: 5
    gauge: 20 AWG
    colors: [BUYE]

connections:
  -
    - X1: [3]
    - Cable 1: [1]
    - X3
  -
    - X1: [1-2]
    - Power Cable: [1-2]
  -
    - Power Cable: [1-2]
    - X2: [1-2]
```

kvid, 2022-02-13:
Thank you for reporting this unexpected SVG output, @norator42 . I'm really sorry for you not receiving any feed-back about this issue for more than a year. That is not intentional, but is probably because the contributors to this project are only able to spend their spare time working with this, and there might be long periods without anyone reading new issues. Then some issues might slip through without any answer when trying to catch up after a low-activity period.

I cannot seem to reproduce your issue using Ubuntu 18.04.4 LTS and WSL1 in Windows 10, Pyhon 3.7.5, and WireViz v[0.2](https://github.com/formatc1702/WireViz/tree/v0.2), v[0.3.2](https://github.com/formatc1702/WireViz/tree/v0.3.2), or the current v[0.4-dev](https://github.com/formatc1702/WireViz/tree/a6efd281248975144f3396f233df84814d8fc911). The older verison 0.1 cannot have been used because your input contains the connector attribute `pinlabels` that was added later.

Can this issue depend on the program rendering the SVG output? I have tried Microsoft Edge Version 97.0.1072.55 (Official build) (64-bit) and Google Chrome Version 97.0.4692.99 (Official Build) (64-bit) in Windows 10. Another factor that might affect the result is the currently installed fonts.

Can someone else try to reproduce this issue in different environments than what I have access to? To avoid such problems, we [encourage everyone to supply enough information to reproduce the reported state](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md) when creating an issue.

JeremyRuhland, 2022-12-17:
I am also experiencing this problem using:
* Ubuntu 22.04.1 
* Python 3.10.6
* wireviz 0.3.2
* graphviz 2.43.0
* graphviz (python module) 0.20.1

I am not experiencing the problem using:
* Ubuntu 18.04.6 LTS
* Python 3.7.5
* wireviz 0.3.2
* graphviz 2.40.1
* graphviz (python module) 0.19.1

Rendered on:
* Firefox 106.0.5 (ubuntu 18)
* Chrome 107.0.5304.110 (ubuntu 18)
* Inkscape 1.0.2 (ubuntu 18)
* Firefox 107.0.1 (ubuntu 22)
* Chromium 108.0.5359.124 (ubuntu 22)
* Inkscape 1.1.2 (0a00cf5339, 2022-02-04) (ubuntu 22)
* Chrome 108.0.5359.124 (windows 10)

I've confirmed that my browsers are actually using arial and not the liberation-serif ubuntu default.

Interesting discovery! Its the svg files, not the browser. The files generated on my 18.04 machine look fine on all my machines, but the ones made on 22.04 exceed borders everywhere.

This issue is not resulting from inside wireviz, I can reproduce it by directly calling dot on the gv file to export the svg and png. The problem goes away if I force dot to render the svg using cairo (this is what happens when pngs are generated), unfortunately this results in large file sizes because each glyph is embedded as a path in the svg and the text is not searchable.

On both systems verbose output from dot confirms that they are using the same path to the same font ttf file. It seems like the newer graphviz (specifically its native svg driver) has a different idea of what arial looks like than the rest of the system (smaller kerning), as such the text boxes it draws are too small for the larger kerning when the font is rendered by the browser. The older version of graphviz correctly estimates the text dimensions so everything lines up.

I think further investigation will have to take place over on [graphviz's gitlab repo](https://gitlab.com/graphviz/graphviz/).

In the meantime my workaround for the issue is to modify `wv_html.py` and replace
https://github.com/formatc1702/WireViz/blob/b0d0070f08247e132b11bf45e617da9e8e1881f1/src/wireviz/wv_html.py#L26-L32
with `file.write(f'[image]\n')`. This of course means the html file is no longer standalone.

ask6483, 2024-04-19:
Hi all,

I have the same issue on Debian testing.
[image]
Tried to change the font to Liberation Sans or DejaVu Sans but the issue remains.

Versions:
* Debian Trixie testing
* graphviz 2.42.2-8
* wireviz 0.4-dev
* python 3.11.8
* python graphviz 0.20.1

kvid, 2024-05-12:
@ask6483 - thank's for confirming this issue. Are you able to include here, one or two image files that demonstrates this problem (together with their YAML source and how they were produced)?

kvid, 2024-05-26:
I suggest adding a few placeholders for usage in HTML templates (a new feature added in v0.4) that can help as a work-around when experiencing SVG problems like this:
- `` = absolute path of output file without extension
- `` = output filename without path or extension
- `` = Base64 encoded PNG diagram that can fit into an image tag like this: `<img src="">`

Then any user can make a custom template using an image tag `<img src="...">` tag for the diagram with either the PNG filename or base64 encoded PNG contents.

Please suggest better names for my suggested placeholders, and maybe other placeholder variations that might help.

ask6483, 2024-05-27:
@kvid sorry for late reply but I lost the track until yesterday notification, here are the files [test_files.zip](https://github.com/wireviz/WireViz/files/15454541/test_files.zip).

It looks like the issue is more visible in html/pdf render than in svg file itself:
1. PDF
[image]
2. SVG
[image]

kvid, 2024-05-28:
@ask6483 - Thank's for the test files. I agree that the issue is greater when rendering your PDF (in Okular v23.08.1 and in Brave v1.65) than when rendering your SVG and HTML (both in Brave v1.65). How did you render your files? And how did you generate the PDF files?

ferdnyc, 2025-06-25:
I took a look at this the other day.

Basically, there are known issues with the way graphviz lays out graphs containing text, particularly in SVG files. It has built-in information (built-in guesstimates, would be more accurate) about the rendered size of each piece of text, and it makes layout decisions based on that info. Problem is, if its measurements are wrong, the sizing doesn't line up. The misalignment varies some from font to font, some from output format to output format (it predicts PostScript layout much better than SVG layout, for example), and most of all from OS to OS, because each one tends to have different ideas of how to render even the same font (or has different versions of the same font).

That being said, I generated `.gv` and `.svg` files for both of the sample WireViz files using both graphviz 2.5.0 and graphviz 12.2.1 on Linux, and the differences start right at the `dot` stage.

`wireviz -fg <input>` generates the `.gv` file the same every time, of course, but when I run it through `dot -Tdot` (which lays out the graph and embeds the computed coordinates into the graphviz source), here's the diff of what that produces with GV 2.5.0 vs. GV 12.2.1:

```diff
--- 1002002/1002002_battery_pack_cable.dot	2025-06-23 00:58:00.520250044 -0400
+++ 1002002.gv12/1002002_battery_pack_cable.dot	2025-06-23 00:59:20.950201249 -0400
@@ -1,5 +1,5 @@
 graph {
-	graph [bb="0,0,1345,231",
+	graph [bb="0,0,1337,237.75",
 		bgcolor="#FFFFFF",
 		fontname=arial,
 		nodesep=0.33,
@@ -19,7 +19,7 @@
 		fontname=arial,
 		style=bold
 	];
-	X1	[height=3.0833,
+	X1	[height=3.1875,
 		label=<
 <table border="0" cellspacing="0" cellpadding="0">
  <tr><td>
@@ -72,10 +72,10 @@
  </td></tr>
 </table>
 >,
-		pos="131.5,111",
+		pos="130.12,114.75",
 		shape=box,
-		width=3.6528];
-	W1	[height=2.5278,
+		width=3.6146];
+	W1	[height=2.6111,
 		label=<
 <table border="0" cellspacing="0" cellpadding="0">
  <tr><td>
@@ -152,16 +152,16 @@
  </td></tr>
 </table>
 >,
-		pos="607.5,140",
+		pos="603.25,143.75",
 		shape=box,
 		style="filled,dashed",
-		width=5.5694];
+		width=5.5278];
 	X1:p1r:e -- W1:w1:w	[color="#000000:#000000:#000000",
-		pos="263,142 327.01,142 342.99,144 407,144"];
+		pos="260.25,146.38 324.25,146.38 340.25,147.75 404.25,147.75"];
 	X1:p2r:e -- W1:w2:w	[color="#000000:#ffff00:#000000",
-		pos="263,119 327,119 343,119 407,119"];
-	X1:p3r:e -- W1:w3:w	[pos="263,95 327,95 343,94 407,94"];
-	B1	[height=2.2361,
+		pos="260.25,122.62 324.25,122.62 340.25,122 404.25,122"];
+	X1:p3r:e -- W1:w3:w	[pos="260.25,98.875 324.26,98.875 340.24,96.25 404.25,96.25"];
+	B1	[height=2.309,
 		label=<
 <table border="0" cellspacing="0" cellpadding="0">
  <tr><td>
@@ -204,12 +204,12 @@
  </td></tr>
 </table>
 >,
-		pos="1148.5,143",
+		pos="1141.6,145.75",
 		shape=box,
-		width=5.4583];
+		width=5.4271];
 	W1:w1:e -- B1:p1l:w	[color="#000000:#000000:#000000",
-		pos="808,144 872.01,144 887.99,142 952,142"];
+		pos="802.25,147.75 866.26,147.75 882.24,145.75 946.25,145.75"];
 	W1:w2:e -- B1:p2l:w	[color="#000000:#ffff00:#000000",
-		pos="808,119 872,119 888,119 952,119"];
-	W1:w3:e -- B1:p3l:w	[pos="808,94 872.01,94 887.99,96 952,96"];
+		pos="802.25,122 866.25,122 882.25,122 946.25,122"];
+	W1:w3:e -- B1:p3l:w	[pos="802.25,96.25 866.26,96.25 882.24,98.25 946.25,98.25"];
 }
```

Basically, GraphViz 12.2.1 renders everything is narrower and taller. The X coordinates are all smaller (shifted to the left), and the Y coordinates tend to be a bit higher (shifted down).

Despite that, at least on Linux there wasn't all that much visible difference in the eventual output, when rendering to SVG. Even with the old version, text tended to overflow the boxes a bit, in SVG... and it doesn't seem that much worse with the newer version than with 2.5.0. But the PNG renders from both versions are spot on. I also ended up changing all of the `arial` fontnames to `Times`, since that was the only PostScript font I could get to render on my system without it getting replaced with Courier, and Evince's renders of those Times-font Postscript files also look good. But the SVGs are at best "eh...", from both versions.

### PNG renders of 1002002, from GV 2.5.0 and 12.2.1

[image]

[image]

### SVG renders of 1002002, again from 2.5.0 and 12.2.1

[image]

[image]

### Since the preceding may look different on your systems than mine, ImageMagick PNG conversions of the previous two files

[image]

[image]

### PostScript versions of the files with `fontname=Times`, as screenshotted from Evince renders (2.5.0 vs. 12.2.1)

[image]

[image]

(I didn't resize the screenshot frame for that last image at all -- the GV 12.2.1 version really does render a bit smaller vertically than the 2.5.0 version. Which is ironic, since as I said the coordinate system was that the image was narrower but taller. But I guess it was LESS taller and MORE narrower, so when resized to the same screen dimensions it came out shorter.)

### Summary

Long and short, as you can see PostScript and PNG lay out perfectly, but GraphViz has issues sizing text in SVG files. Kind of always has, though it may have gotten worse recently on non-Linux systems.

ferdnyc, 2025-06-25:
For the record, the SVG files and the converted PNG renders of those files look virtually IDENTICAL on my system, so that's very representative of the Linux rendering of those SVGs. To the extent the SVGs look DIFFERENT from the PNGs rendered from them, for you, that indicates the difference in how your OS & browser renders SVG files, compared to mine.

ferdnyc, 2025-06-25:
Also FTR, GraphViz can be convinced to do even _worse_ with SVG files, if you use HTML-like labels with any inline styling. Like, the classic GraphViz "that don't look right" graph is as simple as this:

```dot
graph g {
  n [label=<Wow, <font color="blue">some</font> text>];
}
```

The way that renders to SVG on Linux looks fine:

[image]

But if you run that graph through `dot -Tsvg` with any recent Windows Graphviz, you'll likely get an SVG that renders like this (PNG screenshot)

[image]

ask6483, 2025-06-27:
@ferdnyc Can we embed the font in SVG to ensure consistent rendering?

ferdnyc, 2025-06-27:
@kvid Exisiting issue, yes. Several, including https://gitlab.com/graphviz/graphviz/-/issues/1406 which is probably the main one.

Expect a fix... I wouldn't hold my breath, to be perfectly frank. It's a tricky problem. When including flowed text in an SVG, normally either you'd let the size of the enclosing structures be defined by the text boundaries, like in HTML, or you'd flow the text so that it fits within the boundaries. (Either way, the SVG engine would be participating in the layout process.) 

But because GraphViz has its own layout engine, it precisely specifies the position of each element of the diagram, meaning the SVG renderer has no freedom to make adjustments based on font metrics and etc.

(That's what creates the "Wow, sometext" issue on Windows. Because of the font-color change in the middle, graphviz renders "Wow, " "some", and "text" as three separate text spans, and specifies the exact placement of each span based on its own font-size metrics. But its figures don't match the size of the spans as actually rendered by the SVG engine, so each span gets placed progressively farther left of where it should be.

Having to render to PNG is a poor workaround if you really need scalable output. One possible workaround _within_ the SVG space is to just massively increase the `cellpadding` value of table cells with text in them, as graphviz will lay out text consistently in padded cells (meaning any text overflows will no longer escape their cell boundaries).

Here's an SVG generated from that same `1002002_battery_pack_cable` diagram, but with all of the `cellpadding="3"` values WireViz currently uses changed to `cellpadding="10"`:

[image]

And, here's the same change, plus the outer table around the wire passthroughs (only, _not_ any of its inner structure) changed from `cellpadding="0'` to `cellpadding="10"`:

[image]

Unfortunately there's no good way to increase only the _x_ padding of a cell without increasing the _y_ padding, since ideally that's what you'd really want. Embedding `&nbsp;` characters to pad cells horizontally can also work, but then you end up struggling with centering issues and etc...

And, of course, if any of the content uses text-style features like bold, italics, color changes, etc. within the cells, you'll hit the "Wow, sometext" problem even despite the padding.

ferdnyc, 2025-06-27:
@ask6483 Unfortunately that's unlikely to help. If the issue were only about variations in how the SVG diagrams render with different font replacements, then that would solve it. But the coordinates GraphViz uses to lay out the entire diagram are already incorrect ­-- if you embedded the font, it would just mean you'd get _consistent_ overflows in all renderers. But the overflows are already baked into the diagram vectors, it's not a font-**re**placement issue so much as it's a text-_placement_ issue.