wireviz/WireViz #178: [pre-release] Rebuild examples

formatc1702, 2020-10-15:
This is just a reminder to myself to rebuild all examples/demos/tutorials before releasing v0.2.

@contributors:
If there are any other tasks that you feel should be completed before the v0.2 release, please submit them as issues with the `[pre-release]` prefix and I'll add them to the [v0.2 milestone](https://github.com/formatc1702/WireViz/milestone/1) so nothing gets lost.
Once the milestone progress bar shows 100%, I will perform the actual release.

kvid, 2020-10-16:
Your versioning practice seems to be according to accepted recommendations, [PEP 386](https://www.python.org/dev/peps/pep-0386/) and [PEP 396](https://www.python.org/dev/peps/pep-0396/).

formatc1702, 2020-10-16:
[Semantic versioning](https://semver.org/) says:

> How should I deal with revisions in the 0.y.z initial development phase?
>
> The simplest thing to do is start your initial development release at 0.1.0 and then increment the minor version for each subsequent release.

Note that they recommend increasing the *minor* version, not the *patch* version. Since anything before 0.1 is considered "initial development" and anything may change at any time, there is no use in maintaining detailed records using the patch version. Since patch will therefore always be at 0, the conscious decision was made not to include it.
However, as more people use the project, the probability of needing to release quick bugfixes rises, so it *is* worth considering.

> Isn't 0.3.dev regarded as a later version than 0.3?

Similar to `alpha`, `beta` and `rc` versions, adding `dev` to a version places it *before* the version without the suffix.
`1.0.1-alpha` comes before `1.0.1`

I have seen both `.` and `-` as the delimiter before the suffix, but `-` seems to be the more common variant.

kvid, 2020-10-16:
> > Isn't 0.3.dev regarded as a later version than 0.3?
> 
> Similar to `alpha`, `beta` and `rc` versions, adding `dev` to a version places it _before_ the version without the suffix. Semver says:
> 
> > When major, minor, and patch are equal, a pre-release version has lower precedence than a normal version:
> > Example: 1.0.0-alpha < 1.0.0.
> 
> I have seen both `.` and `-` as the delimiter before the suffix, but `-` seems to be the more common variant.

You are right. Personally, I think it's easier to accept  0.3-dev < 0.3 with a dash instead of a dot, but others might disagree.

It seems the conflict issue in #172 never made it into v0.2 after all - I hope that's your intention and not just forgotten.

formatc1702, 2020-10-16:
I can see this issue really has you worried ;)

The way I see it, the current setup works, and will not cause any problems with the release.
I am leaving the issue open to hopefully get some clarity in the future, but I will not let this delay the release any further.
I was tempted to at least remove the lone `.` but I'd rather wait for definitive confirmation on that too.
The duplicate requirements list is, especially with a total of only three packages, mainly an issue of purism, not of functionality or maintainability at this point.

So, not forgotten, but deliberately left open to avoid multiple steps of fixing without a definitive goal.

---

> Personally, I think it's easier to accept 0.3-dev < 0.3 with a dash instead of a dot, but others might disagree.

I've changed it to `-` because I feel the same after re-reading the SemVer docs.

formatc1702, 2020-10-17:
v0.2 has been released! Thanks everybody for your contributions.

I will start going through the PRs as discussed in #101 in the following days.

closed 2020-10-17 by formatc1702