wireviz/WireViz #182: [feature]Add ability to customize HTML output title

martonmiklos, 2020-10-19:
I think it would be beneficial to being able to generate the title of the HTML output. 
I often keep multiple windows/tabs containing Wirewiz generated outputs open and now I cannot distinguish easily between them:

[image]

Considered as a low hanging fruit to me in the case if noone has objections. I am also open for the syntax recommentations!

kvid, 2020-10-19:
This has already been suggested in https://github.com/formatc1702/WireViz/issues/158#issuecomment-687924335. Feel free to join the syntax discussion in #158.

formatc1702, 2020-10-20:
@martonmiklos if you feel that #158 is enough to address this topic (and the more general metadata question), please close this issue. If you feel it needs to remain open, please let me know why. No problem either way.

martonmiklos, 2020-10-20:
Hi folks, my bad, but the github issue search did not resulted that issue by searching for 'title'.

Closing it, having it covered by the meta syntax is perfectly fine.

closed 2020-10-20 by martonmiklos