wireviz/WireViz #183: [bug] Bundles to have consistent part number fields

Tyler-Ward, 2020-10-20:
Splitting out of https://github.com/formatc1702/WireViz/pull/115#discussion_r508000264

Curently in a bundle the part number fields will be rendered differently if they are a string or a list. If they are a string they are rendered at the top of a node and if a list they are rendered per wire, it mould make sense for this to be consistent across bundles by always doing the latter.

kvid, 2020-10-20:
AFAIK, this is by design. Attributes common to all wires are listed above all wires (except image, and notes below), and attributes specified for each wire listed above and below each wire. Currently, both options are implemented for `pn`, `manufacturer`, and `mpn`, but it has been discussed to support the same for `gauge` too.

formatc1702, 2020-10-21:
> AFAIK, this is by design.

Correct. Here's the history:
- At the beginning, the only parameter that could change on a per-wire basis, was color. Everything else (type, length, gauge, ...) was global for the entire cable, thus displayed above the wires.
- Currently, bundles implement separate [manufacturer] part number parameters.
- In the future (see #56), more parameters might be editable on a per-wire basis.

That's why, IMHO, we shouldn't see this as a bug, nor worry about it too much, because I think some major refactoring may be in order to cleanly implement the transition for more granularity.